chipsalliance/chisel #951: Porting black boxes from Chisel 2 to Chisel 3

schoeberl, 2018-11-23:
**Type of issue**: bug report

Is there a document on how black boxes changed from Chisel 2 to Chisel 3? I get errors that setName is not a member of UInt/Bool... and setModuleName, renameClock is not found

**What is the use case for changing the behavior?**

Porting Patmos from Chisel 2 to Chisel 3

edwardcwang, 2018-11-23:
Can you share a link to the Blackbox in question? One option that comes to mind is `ExtModule`.

edwardcwang, 2018-11-23:
Hmm, this looks like #612.

One option for `EthMacBB` could be as follows (untested). Essentially you use an `ExtModule` to have fine grained control over the IOs and blackbox name and then create a wrapper around it that maps those names to the Chisel bundles.

```scala
class eth_controller_top(val extAddrWidth: Int, val dataWidth: Int) extends ExtModule(Map("BUFF_ADDR_WIDTH" -> extAddrWidth.toString)) {
  val clk = IO(Input(Clock()))
  val rst = IO(Input(Bool()))

  val MCmd = IO(Output(UInt(3.W)))
  val MAddr = IO(Output(UInt(extAddrWidth.W)))
  val MData = IO(Output(UInt(dataWidth.W)))
  val MByteEn = IO(Output(UInt(3.W)))

  val SData = IO(Input(UInt(dataWidth.W)))
  val Resp = IO(Input(UInt(2.W)))
  
  // mtx_clk_pad_i etc...
}

class EthMacBB(val extAddrWidth: Int = 32, val dataWidth: Int = 32) extends Module {
  val io = new OcpCoreSlavePort(extAddrWidth, dataWidth) with EthMac.Pins

  val bb = Module(new eth_controller_top(extAddrWidth=extAddrWidth, dataWidth=dataWidth))
  
  bb.clk := clock
  bb.rst := reset

  io.M.Cmd := bb.MCmd
  io.M.Addr := bb.MAddr
  io.M.Data := bb.MData
  io.M.ByteEn := bb.MByteEn
  bb.SResp := io.S.Resp
  bb.SData := io.S.Data

  // mtx_clk_pad_i etc...

  // keep some sigals for emulation
  debug(io.M.Cmd)
  debug(io.M.Addr)
  debug(io.M.Data)

  // registers to help emulation
  val respReg = Reg(UInt(2.W))
  val dataReg = Reg(UInt(dataWidth.W))
  io.S.Resp := respReg
  io.S.Data := dataReg
}
```

schoeberl, 2018-11-23:
Thanks for the proposal to use chisel3.experimental.ExtModule.

Looks like it more or less compiles. I’ve just one error with the Map parameter, maybe unrelated.

[error]  found   : String
[error]  required: chisel3.core.Param
[error] class eth_controller_top(val extAddrWidth: Int, val dataWidth: Int) extends chisel3.experimental.ExtModule(Map("BUFF_ADDR_WIDTH" -> extAddrWidth.toString)) {
[error]                                                                                                                                                  ^

Otherwise this gives me a little headache for a smooth transition form Chisel 2 to 3, as this change would not work in Chisel 2. But anyway, I have just 9 black boxes in the whole design ;-)

Cheers,
Martin

edwardcwang, 2018-11-23:
Does `Map("BUFF_ADDR_WIDTH" -> IntParam(extAddrWidth))` work?

schoeberl, 2018-11-23:
Yes, that works.

Thanks!
Martin

schoeberl, 2018-11-23:
If there is no plan to support Chisel 2 compatible BlackBoxes as they where, then yes this can be closed.

Martin

closed 2018-11-23 by schoeberl