chipsalliance/chisel #954: Can assign registers from different clock domain

ducky64, 2018-11-30:
It's possible to assign a register in a different clock domain than the one it was created in, but the assignment still happens on the register instantiation clock. For example:
```scala
val a = Reg(...)
withClock (someDifferentClock) {
  a := somethingElse
}
```

But a still updates on the enclosing clock, not `someDifferentClock`.

The register assignment in a different clock domain should probably be an error, as the results are not what would be expected from reading the code.

**Type of issue**: other enhancement

**Impact**: API modification

**Development Phase**: proposal

aswaterman, 2018-11-30:
When doing explicit clock gating, we tend to wrap most of the module code inside a single withClock. The handful of registers that aren’t gated are declared outside the withClock, but their assignments (along with all the rest of the code) are inside the withClock.

Here’s one example: https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/rocket/RocketCore.scala#L879 is the withClock, and https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/rocket/RocketCore.scala#L98 are the register declarations.

edwardcwang, 2018-11-30:
Would second @aswaterman (being able to do `val altClockReg = withClockAndReset (altClk, altRst) { RegInit(0.U(16.W) }` is useful and sound from a hardware construction perspective).

From a hardware construction perspective this makes sense, since you're plopping down a register and connecting its clk and reset ports as specified. It doesn't make sense to have a register update on multiple clocks since that's not what flip-flop cells do.

Even in Verilog (where one can accidentally write this), this fails in synthesis:

https://stackoverflow.com/questions/32729676/how-to-use-two-events-in-an-always-block-in-verilog
https://electronics.stackexchange.com/questions/365470/sharing-data-in-two-different-always-posedge-blocks

ducky64, 2018-12-15:
Resolution from the dev meeting today was "working as intended".

There is also a proposed feature to check unsafe clock domain crossings in FIRRTL (with safe crossings as an annotation), though it not really the same issue (since reconstructing lexical scope information like withClock scopes in FIRRTL is iffy at best).

closed 2018-12-15 by ducky64