chipsalliance/chisel #955: Support context types in Bundle

tampler, 2018-12-02:
**Type**: Feature Request 
**Description**: I'd like to abstract construction of some types and parametrize them depending on **application context**. Particularly, in DspTools project, I need the following:
```scala
class SimpleComplexAdder extends Module {
  val io = IO(new Bundle {
    val a1 = Input(DspComplex(FixedPoint(6.W, 4.BP), FixedPoint(6.W, 4.BP)))
    val a2 = Input(DspComplex(FixedPoint(8.W, 1.BP), FixedPoint(8.W, 1.BP)))
    val c  = Output(DspComplex(FixedPoint(14.W, 5.BP), FixedPoint(14.W, 5.BP)))
  })
```
The same hardware needs to support another types as well
```scala
class SimpleComplexAdder extends Module {
  val io = IO(new Bundle {
    val a1 = Input( FixedPoint(6.W, 4.BP))
    val a2 = Input(FixedPoint(8.W, 1.BP))
    val c  = Output(FixedPoint(14.W, 5.BP))
  })
```
The above data types can't be parametrized with a single parameter like 
```scala
class SimpleComplexAdder[A <: Data] (dType:A) extends Module { ... }
```
 since DspComplex type requires two inputs and FixedPoint requires only one. However, both data types express Monadic properties and have exactly the same operations in their corresponding (operation) Rings.

So both of them can be abstracted with a **Context**, similar to Either, Option, Future, etc in Functional programming. The only thing to implement is a **data context**, which puts every input in a corresponding hardware context variable.

DspTools guys offered a solution, implementing own Context for each type, which is verbose and requires a lot of boilerplate. 

**My possible solution** for this is here:
https://github.com/Neurodyne/chat/blob/2c03e439cd697ffbee0aaec1bb95417266008540/src/main/scala/AbstractIO.scala#L6

However, this won't compile, yielding in: 

[error] AbstractIO.scala:9:14: inferred type arguments [C[A]] do not conform to method apply's type parameter bounds [T <: chisel3.core.Data]
[error]     val a  = Input (inType)

Implementing such feature will simplify Diplomacy code and parametrization of complex hardware units in Rocket and Boom

chick, 2018-12-04:
Input() and Output() require their arguments to be of type Data. So the first thing you need to do to keep moving forward here is to add
```scala
class AbstractIO[C[_] <: Data, A <: Data, B <: Data] ( inType:C[A], outType:C[B]) extends Module {
```

tampler, 2019-01-18:
I was actually able to do the job without type constructors. 

@chick The suggestd implementation won't actually work, since C[_] is a type constructor ( List, Map, Seq ), not a data type.
```scala
class AbstractIO [C[_] <: Data, A <: Data, B <: Data]
```
The proposed solution is here: https://github.com/Neurodyne/chisel_katas/blob/20c91a6781ee39fd827296a414a517ab9a5dda03/src/main/scala/Ring.scala#L73

Works with UInt, SInt, FixedPoint[UInt], Complex [SInt,SInt], Complex [FixedPoint[SInt], FixedPoint[SInt]], etc.

Published as a Chisel kata for the benefit of newcomers

tampler, 2019-01-22:
Seems to work okay. Thanks for your feedback!

closed 2019-01-22 by tampler