chipsalliance/chisel #957: [RFC] Bits as full Types

schoeberl, 2018-12-08:
Bits in Chisel are currently second-class citizens. This is different from Chisel 2, where more functions with Bits are available. Therefore, the reduced functionality of Bits creates a compatibility issue with Chisel 2.

Besides that, Bits as type has a merit on its own (see below).

**Type of issue**: feature request

**Impact**: API addition (no impact on existing Chisel 3 code), but improves Chisel 2 compatibility

**Development Phase**: proposal

Not all data types in hardware are signed or unsigned integers. For example, a word in memory may be an integer, an instruction, or a floating point values. Or something else. Representing this word as UInt is not the correct type. The same is true for a register in a processor, the data field in a bus, and more.

For stronger typing, it makes sense to have a complete Bits type. Bits shall be a collection of bits without an arithmetic meaning. Allowed operations are logical functions (&, |, ^, ~), reduction, and comparison (===, =/=). There is no place for addition or < comparison. Bits can be cast to an integer type.

One question is, how bit vectors with different widths are handled. Either automatic expansion as for UInt or having the length as part of the Bits type and disallow operations with different lengths. The later would be a stronger type restriction. If easy to implement, the later is preferable.

When looking at VHDL as an example language, there are three types: std_logic_vectore, which has no arithmetic operations, unsigned, and signed. Bits would be the equivalent of std_logic_vector.

Bits can be lowered to UInt in FIRRTL and then Verilog without losing the additional type safety in Chisel 3.

ducky64, 2018-12-15:
This was discussed at the dev meeting today, though as @schoeberl was not present, we didn't make much progress or finalize any resolutions.

Concerns mainly revolved around the details, especially what happens when implicit width expansion is needed. Compatibility with current chisel3 code is also a concern, since the `Bits(...)` companion object constructor (perhaps counterintuitively) returns type UInt, and having it return Bits may break code.

azidar, 2019-01-18:
Make a PR, have a deprecation schedule for the arithmetic operators on Bits.

kammoh, 2019-07-29:
I'm experimenting on promoting `Bits` as a full concrete type in #1133 along with other clean-ups of the type system. Everything seems to be in place now, while seems not to be breaking the currently deployed code-base (at least the ones I was I able to verify). 
We can possibly add a factory object for the concrete `Bits` type with a new name and deprecate the current `Bits` factory object producing `UInt`.
Being part of the API brings some challenges including finding a suitable name for the factory object (Bitvector? PureBits? LogicBits? ...) and also facing the potential (interim) confusion arising while a factory named differently that Bits is generating `Bits` objects.
Enforcing compile-time type safety for comparison operators (`===`, `=/=`) while also defined for `Bits` would probably be not feasible if we make the current implementation of `Bits` (as is the superclass for arithmetic types) directly a concrete class.
The concrete type could possibly be introduced as a new sub-type of bits (solves interim naming discrepancy, type-safe comparison operators,  and also the possibility of having operations only defined for the new "pure" Bits type and not e.g. for `SInt`)