chipsalliance/chisel #958: RESET

myuanqing, 2018-12-08:
How can I use the RESET port?
I have found that in the verilog code, there is a reset input,
[image]
which I didn't define in the chisel code.

How can I use this port in the chisel code?

edwardcwang, 2018-12-08:
The `reset` port is implicit and automatically used by all registers in the module. You can reference it explicitly by using `reset`. e.g. `foobar.io.reset_2 := reset`.

ducky64, 2018-12-08:
Also, you can use `chisel3.experimental.withClockAndReset { ... }` (or `withClock` or `withReset`) to establish a clock and/or reset scope.

Note that resets are synchronous, though async reset is a planned feature.

myuanqing, 2018-12-08:
I see. Thank you for your answers.
But how can I use the reset port in testing code?
I have tried this:
[image]
but it couldn't work.

edwardcwang, 2018-12-08:
In general `reset` and `clock` are special signals (at least in current chisel-testers) and cannot be `poke`d like normal signals. `clock` can be stepped with `step(1)` and `reset` is automatically toggled at the beginning of a test.

myuanqing, 2018-12-09:
Okay. Thanks a lot.

closed 2018-12-09 by myuanqing