chipsalliance/chisel #959: [RFC] Rename WireInit to WireDefault

schoeberl, 2018-12-10:
When describing combinational circuits a common error is not covering all paths with an assignment. This can happen in Verilog and in VHDL, not in Chisel (which is very good!). To avoid this error common practice is to start with a *default* assignment for the signal. Chisel WireInit does the same. However, the name Init is misleading (relative to RegInit) as it sounds like something is set at initialization time (at reset). A default assignment "executed" alsways. Therefore, I would propose to rename WireInit to WireDefault for better readability.

**Type of issue**: feature request

**Impact**:  API modification

aswaterman, 2018-12-10:
I have to agree that WireDefault is a clearer name, as unlike RegInit it is unrelated to the concept of reset.  Obviously for compatibility we'd need to keep WireInit, though.

schoeberl, 2018-12-11:
WireInit can immediately be marked as depreciated. Google found just a single file where it is used: https://github.com/ucb-bar/dsptools/blob/master/rocket/src/main/scala/dspblocks/MemToRegmap.scala

Cheers,
Martin

ducky64, 2018-12-15:
This was discussed at the dev meeting today. There's agreement that WireDefault is better terminology, though it's unclear whether the impact of even a deprecation phase is worth the gain - especially if hundreds of deprecated warnings start showing up. If we don't decide to change WireInit to WireDefault, one alternative is to give WireInit better documentation, as a wire with an initial assignment.

A code rewriter may mitigate the impact of deprecations.

Note: VecInit may be in a similar situation.

Historical note: WireInit was a refactoring of the Wire(init=...) construct.

aswaterman, 2018-12-15:
I know my opinion on the matter isn't shared globally, but we don't actually have to deprecate old interfaces.  Could just add WireDefault as an alias for WireInit.  Having only one way to do things is a guideline, not a dogma.

ducky64, 2018-12-15:
I think the disadvantage to having multiple interfaces that do the same thing is that there's more to learn. Obviously not a problem for you, me, or those who follow these issues - we know what it does - but for someone new who might be learning Chisel from examples, it could be detrimental.

I don't think the benefit of the change outweighs the drawbacks of having two ways to do the same thing.

That being said, I'm not entirely opposed to keeping both around. The main concern would be that at least new users should be aware of the recommended way - so that means some kind of warning (or maybe something more clever?) - though it'd have to be opt-out instead of opt-in because I don't think it's practical to rely on people finding and adhering to style guides.