chipsalliance/chisel #960: We really want this to happen!

merencelik, 2018-12-11:
Please fill in the gaps:

Windows
TODO: write me. If you really want to see this happen, let us know by filing a bug report!

seldridge, 2018-12-11:
Thanks for checking out Chisel and for filing an issue about the stub [`Windows` section of the README](https://github.com/freechipsproject/chisel3#windows).

As this is written, it sounds like Windows is unsupported, but this is incorrect... While these are famous last words, Chisel should work just fine on Windows. Once you have `sbt` installed (https://www.scala-sbt.org/1.x/docs/Installing-sbt-on-Windows.html) you should be do Chisel development in this repo or build out a project (with [`chisel-template`](https://github.com/freechipsproject/chisel-template) being a better repo to start from). As far as I'm aware, the known issues with Windows relate to Verilator, specifically some of the problems outlined here: https://github.com/freechipsproject/chisel3/issues/874. 

I do not have access to any Windows machines, so some of the other devs can see about expanding the instructions here. However, you're more than welcome to take a stab at that.

ducky64, 2018-12-11:
I do regular development in chisel3 on Windows, and as has been said before, everything but Verilator works fine. The practical implication is that you won't be able to run the chisel3 regressions natively, but in general for simulation you can:
- run Verilator in WSL or Cygwin. I've run the Chisel regression suite in WSL, and it works fine.
- for pure Chisel code, you can use treadle, a Scala-based FIRRTL simulator. There is also some support for writing Scala models of blackboxes, eg AsyncResetReg. You'll either need to use the chisel-testers (http://github.com/ucb-bar/chisel-testers) or testers2 (http://github.com/ucb-bar/chisel-testers2), since chisel3 base currently doesn't have support for anything but Verilator.

Note that you'll be able to generate Verilog from Chisel just fine, which can be pushed to downstream (eg FPGA) tools.

As for this issue in particular, we should probably add instructions to the main README, I think it's mainly just 'download and install SBT'. And a discussion of limitations and trade-offs.

PR #964 (2018-12-24): Add Windows setup instructions to readme
Title is self-explanatory.

**Related issue**: Fixes #960 

**Type of change**: other enhancement

**Impact**: no functional change

**Development Phase**: implementation
closed 2018-12-24 by ducky64