chipsalliance/chisel #965: Failure when comparing against large UInt literal

hutch31, 2018-12-19:
bug report

**Impact**: unknown

**Other information**
 

Comparing a 32-bit UInt against a large literal results in error message.

val x = UInt(32.W)
when (x === 0xFFFFFFFF.U) {
  ...
}

Results in:
[error] (run-main-0) java.lang.IllegalArgumentException: requirement failed: UInt literal -1 is negative
[error] java.lang.IllegalArgumentException: requirement failed: UInt literal -1 is negative

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
See attached testcase to reproduce the issue

**What is the current behavior?**
Fails with error message

**What is the expected behavior?**
Process the literal as an unsigned constant rather than a twos-complement constant

**What is the use case for changing the behavior?**
Awkward to create literals without being able to express them directly

[chisel_uint_bug.zip](https://github.com/freechipsproject/chisel3/files/2692816/chisel_uint_bug.zip)

seldridge, 2018-12-19:
This is a Scala/Java thing. `0xFFFFFFFF` is an `Int`, so this can't be anything other than `-1`. E.g., trying that on the Scala REPL:
```
scala> 0xFFFFFFFF
res0: Int = -1
```

However! You can do this with either a `String` or a `BigInt`. I think both of the following work:

```scala
val x = UInt(32.W)
when (x === "hffffffff".U || x === BigInt("ffffffff", 16)) {
  \\...
}
```

ducky64, 2018-12-19:
Some additional notes:
1. You can also use a long, `0xffffffffL.U`, though this will break at 64 bits.
2. For the String-to-UInt conversion, `x` also works as a hex radix specifier, eg `"xffffffff".U`

This isn't the first time we've seen this, so it might be worth adding a helpful "did you mean to ..." message for this type of error...

hutch31, 2018-12-19:
Thanks, string to UInt much cleaner than 
Fill(32, 1.asUInt(1.W))

closed 2022-02-28 by hutch31