chipsalliance/chisel #968: loadMemoryFromFile doesn't work with memories of aggregate types

nhynes, 2018-12-26:
**Type of issue**: bug report

**Impact**: API addition (no impact on existing code)

**What is the current behavior?**

`Mem` of type `Bundle` is [split](https:/github.com/github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/passes/LowerTypes.scala#L69) into single-type memories with names like `theMem_field1` and `theMem_field2`. Unfortunately, the [`loadMemoryFromFile` annotation](https:/github.com/github.com/freechipsproject/chisel3/blob/master/src/main/scala/chisel3/util/experimental/LoadMemoryTransform.scala#L41) only contains the top-level name, so when [processed by treadle](https:/github.com/github.com/freechipsproject/treadle/blob/master/src/main/scala/treadle/executable/Memory.scala#L614), one receives the error
```
java.util.NoSuchElementException: key not found: MyModule.theMem
```

I'm not quite sure how to insert the fix (hence the issue rather than a PR), but I think that it involves having chisel creating several annotations since [not much is done to the annotation once it reaches firrtl](https:/github.com/github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/annotations/LoadMemoryAnnotation.scala#L21).

**What is the expected behavior?**

`loadMemoryFromFile` should work on memories of aggregate types.

**Please tell us about your environment:**

`master` branch of all relevant repos

nhynes, 2018-12-27:
Okay, so I'm going to try to fix this by adding a firrtl pass which replaces a `LoadMemory` annotation for a `Bundle` memories with one `LoadMemory` for each sub-data.

The fix should definitely happen at the firrtl level since chisel doesn't know that memories are split; and both chisel2verilog and treadle depend on the firrtl annos (meaning that there's a common place to insert the modification).

Writing a compiler pass is motivated by annotations generally not having type information. To add this would also require chisel annotations to know too much about the firrtl IR (i.e. `fir.Type`). Thus, the only place where types and annotations are seen together is in `CircuitState`, which is only available to passes.

nhynes, 2018-12-27:
Looks like the actual issue is that the treadle `ExecutionEngine` does another lowering pass but with the original (i.e. non-renamed) `LoadMemoryAnnotation`s. The `ToLoFirrtl` seems to be necessary, though, since the input appears to be HighFirrtl.

This is a temporary fix:

```diff
--- a/src/main/scala/treadle/executable/ExecutionEngine.scala
+++ b/src/main/scala/treadle/executable/ExecutionEngine.scala
@@ -422,6 +423,9 @@ object ExecutionEngine {
     val timer = new Timer

     val loweredAst: Circuit = if(treadleOptions.lowCompileAtLoad) {
+      optionsManager.firrtlOptions = optionsManager.firrtlOptions.copy(
+        annotations = optionsManager.firrtlOptions.annotations.filter(_.isInstanceOf[LoadMemoryAnnotation])
+      )
       if(treadleOptions.allowCycles) {
         optionsManager.firrtlOptions = optionsManager.firrtlOptions.copy(
           annotations = optionsManager.firrtlOptions.annotations :+ DontCheckCombLoopsAnnotation
```

nhynes, 2018-12-27:
Turns out the real issue was that `chisel-testers2` was passing the old, un-renamed annotations into the `ToLoFirrtl` pass. This is what I get for using bleeding edge repos!

closed 2018-12-27 by nhynes

jackkoenig, 2018-12-28:
Thanks for finding and fixing the bug :)