chipsalliance/chisel #971: Incorrect module name for nested LoadMemoryAnnotation

nhynes, 2018-12-28:
**Type of issue**: bug report

**Impact**: no functional change

**What is the current behavior?**

`LoadMemoryAnnotation` records the `ComponentName` of the target memory, so treadle looks for the memory as `ModuleName.memoryField`. However, when this annotation isn't the Top module, the FIRRTL has the memory as `instanceName.memoryField`, which leads to an error when initializing the memory.

**Pseudo-MWE**

```scala
class MyModule extends Module {
  val io = IO(...)
  val mem = SyncReadMem(42, Bool())
}

class Top extends Module {
  val io = IO(...)
  val mm = Module(new MyModule)
}

// in TopTest.scala
class TestingTop extends Top {
  loadMemoryFromFile(mm.mem, "memfile")
  // annotation could also go in MyModule itself but gives the same error
}
```

Compiling this yields the following error

```
java.util.NoSuchElementException: key not found: MyModule.mem
...
...
at treadle.executable.MemoryInitializer.$anonfun$memoryMetadata$1(Memory.scala:614)
```

since, in fact, the treadle `SymbolTable` (which is collected directly from the low firrtl) contains `mm.mem` instead.

**What is the expected behavior?**

It should Just Work (TM).

As ever, I have no idea why this bug is happening, but it's keeping me from writing this test, so I must plough through and figure it out!

**Please tell us about your environment:**

`master` branch of all relevant repos (including `chisel-testers2` since that seems to be relevant)

nhynes, 2018-12-28:
Yep. DCE FTW. Sorry for the spam.

closed 2018-12-28 by nhynes

Croyyin, 2021-06-15:
> Yep. DCE FTW. Sorry for the spam.
> 
> For reference, a quick and dirty fix is to just
> 
> ```scala
> ptionsManager.firrtlOptions.annotations :+ firrtl.transforms.NoDCEAnnotation
> ```

I'm sorry to disturb you, but I wonder where I should use ''ptionsManager.firrtlOptions.annotations :+ firrtl.transforms.NoDCEAnnotation''.

wubinyi, 2021-07-02:
> I'm sorry to disturb you, but I wonder where I should use ''ptionsManager.firrtlOptions.annotations :+ firrtl.transforms.NoDCEAnnotation''.

You can do it like the following code.

import org.scalatest._
import chiseltest._
import chisel3._
import chiseltest.experimental.TestOptionBuilder.ChiselScalatestOptionBuilder
import chiseltest.internal.TreadleBackendAnnotation
import firrtl.transforms.NoDCEAnnotation

class myModuleTests extends FlatSpec with ChiselScalatestTester with Matchers {
	behavior of "my module"
	
	it should("do something") in {
		val annotations = Seq(
			TreadleBackendAnnotation,
			NoDCEAnnotation
		)
		test(new MyModule()).withAnnotations(annotations) {dut =>
                        // "write your test"
		}
	}
}