chipsalliance/chisel #976: BlackBox parametrization

tampler, 2019-01-06:
Wanna extend a BBox example, but have no write permission. So pls, integrate this.

1. Verilog and SV standard (IEEE2009) won't allow String and Real parameters in a synthesizable code. So, pls remove the following from Chise3/*/BlackBox.scala
```scala
case class DoubleParam(value: Double) extends Param
case class StringParam(value: String) extends Param
/** Unquoted String */
case class RawParam(value: String) extends Param
```

2. Put an example of BBox parametrization like below:
```scala
class BBAdd (dwin: Int, dwout: Int) extends BlackBox ( 
Map ("DWIN" -> IntParam(16), "DWOUT" -> IntParam(32), "PIPE" -> IntParam(0)))
with HasBlackBoxResource {...}

```

edwardcwang, 2019-01-06:
1) Can you show us where it says that?

2) There is already an example of parametrization on the BlackBox page on the wiki: https://github.com/freechipsproject/chisel3/wiki/BlackBoxes

tampler, 2019-01-06:
1. /chisel3/chiselFrontend/src/main/scala/chisel3/core/BlackBox.scala
2. I saw that. This parametrization won't work for Ints. You need to wrap it in the **IntParam** class to compile. Newcomers may not see that, so add to wiki.

Real and String port parametrization is forbidden in verilog. So remove this

closed 2019-03-23 by tampler