chipsalliance/chisel #989: getWidth not working on hardware generated by "asUInt" and parameterized IO bundles.

stevenmburns, 2019-01-17:
**Type of issue**: bug report

**Other information**
getWidth fails (Width of chisel3.core.UInt@1303ce95 is unknown!) in cases where we think it should be able to determine the width. 
In the code below, Bad0 is a RegInit of a value constructed using asUInt. Bad1 is similar, but uses WireInit instead of RegInit. Bad2 is the case of an input wire whose width is determined by a parameter.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
The following chisel code demonstrates the problem(s):
````
package width_bug

import chisel3._

class Ifc(num: Int) extends Module {
    val io = IO(new Bundle {
        val in = Input(UInt(num.W))
        val out = Output(UInt(num.W))
    })
}

// these fail only if getWidth is needed during elaboration such as in the require statement below
class Bad0( num: Int) extends Ifc(num) {
    val something = RegInit(VecInit(Seq.fill(num)(true.B)).asUInt)
    require(something.getWidth == num)
    io.out := something
}

class Ok0( num: Int) extends Ifc(num) {
    val something = RegInit(VecInit(Seq.fill(num)(true.B)).asUInt)
    io.out := something
}

class Bad1( num: Int) extends Ifc(num) {
    val something = WireInit(VecInit(Seq.fill(num)(true.B)).asUInt)
    require(something.getWidth == num)
    io.out := something
}

class Ok1( num: Int) extends Ifc(num) {
    val something = WireInit(VecInit(Seq.fill(num)(true.B)).asUInt)
    io.out := something
}

class Bad2( num: Int) extends Ifc(num) {
    val something = WireInit(io.in)
    require(something.getWidth == num)
    io.out := something
}

class Ok2( num: Int) extends Ifc(num) {
    val something = WireInit(io.in)
    io.out := something
}
    
// these pass 
class Good0( num: Int) extends Ifc(num) {
    val s = 3.U(num.W)
    val something = RegInit(s)
    require(something.getWidth == num)
    io.out := something
}

class Good1( num: Int) extends Ifc(num) {
    val something = VecInit(Seq.fill(num)(true.B)).asUInt
    require(something.getWidth == num)
    io.out := something
}

class Good2( num: Int) extends Ifc(num) {
    val something = WireInit(2.U(num.W))
    require(something.getWidth == num)
    io.out := something
}

class Good3( num: Int) extends Ifc(num) {
    val something = RegInit(("b" + "1" * num).U)
    require(something.getWidth == num)
    io.out := something
}

object Bad0 {
     def main(args: Array[String]): Unit = {
        chisel3.Driver.execute(args, () => new Bad0(5))
    }
}

object Bad1 {
     def main(args: Array[String]): Unit = {
        chisel3.Driver.execute(args, () => new Bad1(5))
    }
}

object Bad2 {
     def main(args: Array[String]): Unit = {
        chisel3.Driver.execute(args, () => new Bad2(5))
    }
}

object Ok0 {
     def main(args: Array[String]): Unit = {
        chisel3.Driver.execute(args, () => new Ok0(5))
    }
}

object Ok1 {
     def main(args: Array[String]): Unit = {
        chisel3.Driver.execute(args, () => new Ok1(5))
    }
}

object Ok2 {
     def main(args: Array[String]): Unit = {
        chisel3.Driver.execute(args, () => new Ok2(5))
    }
}

object Good0 {
     def main(args: Array[String]): Unit = {
        chisel3.Driver.execute(args, () => new Good0(5))
    }
}

object Good1 {
     def main(args: Array[String]): Unit = {
        chisel3.Driver.execute(args, () => new Good1(5))
    }
}

object Good2 {
     def main(args: Array[String]): Unit = {
        chisel3.Driver.execute(args, () => new Good2(5))
    }
}

object Good3 {
     def main(args: Array[String]): Unit = {
        chisel3.Driver.execute(args, () => new Good3(5))
    }
}
````

**What is the current behavior?**
Fails with a chisel exception.

**What is the expected behavior?**
Compute the correct value of the width.

**Please tell us about your environment:**
Recent version of Chisel/FIRRTL from top of trunk.

**What is the use case for changing the behavior?**
We use getWidth to determine how to construct other parts of the circuit.

ucbjrl, 2019-01-18:
For the `RegInit` case, it looks like this is a result of:
```scala
  /** @usecase def apply[T <: Data](init: T): T
    *   Construct a [[Reg]] initialized on reset to the specified value.
    *   @param init Initial value that serves as a type template and reset value
    */
  def apply[T <: Data](init: T)(implicit sourceInfo: SourceInfo, compileOptions: CompileOptions): T = {
    val model = (init match {
      // If init is a literal without forced width OR any non-literal, let width be inferred
      case init: Bits if !init.litIsForcedWidth.getOrElse(false) => init.cloneTypeWidth(Width())
      case init => init.cloneTypeFull
    }).asInstanceOf[T]
    RegInit(model, init)
  }
```
I'm not sure why the condition isn't:
```scala
      // If init is a literal without forced width OR any non-literal with unknown width, let width be inferred
      case init: Bits if !(init.litIsForcedWidth.getOrElse(false) || (!init.litOption.isDefined && init.isWidthKnown)) => init.cloneTypeWidth(Width())
```

aswaterman, 2019-01-18:
@ucbjrl  I think it’s mostly just historical but it will absolutely break some people’s designs in hard to detect ways if this behavior were changed at this point.

In general, if you want getWidth to work on a Reg, you should explicitly set its width (or initialize it to a literal of fixed width).

jackkoenig, 2019-01-18:
In fact a prior change of this behavior **did** break our code in a subtle way that was only found in simulation: https://github.com/freechipsproject/chisel3/pull/952

stevenmburns, 2019-01-18:
Perhaps I just being stupid here, but how is making something return a correct value instead of asserting going to break anyone's existing code. I have not tried this on something with #952 applied. I'll look at that and make another comment.

aswaterman, 2019-01-18:
@stevenmburns it changes the semantics of existing working code because it changes the width-inference behavior for registers.  In general, if Reg.getWidth throws an exception, it means that a subsequent assignment to that register could increase its width.  Any change that makes Reg.getWidth work in more situations has the side effect of changing the width-inference behavior for that same set of situations.

stevenmburns, 2019-01-23:
I guess the more interesting issue is that the following two WireInit invocations produce different results.
We'll just use the appropriate one in the case we want:
````
package width_bug

import chisel3._
import chisel3.iotesters.{ChiselFlatSpec, Driver, PeekPokeTester}

class Ifc extends Module {
    val io = IO(new Bundle {
        val en = Input(Bool())
        val in = Input(UInt(2.W))
        val out = Output(UInt(3.W))
    })
}

class Tester extends ChiselFlatSpec {

   "Explicit type" should s"not expand" in {
      Driver(() => new Ifc {
        val w = WireInit( io.in.cloneType, init=io.in)
        when ( io.en) {
          w := 7.U
        }
        io.out := w
      }, "treadle") {
        c => new PeekPokeTester(c) {
	  poke( c.io.en, 1)
	  expect( c.io.out, 3)
	}
      } should be (true)
    }

   "Implicit type" should s"expand" in {
      Driver(() => new Ifc {
        val w = WireInit( io.in)
        when ( io.en) {
          w := 7.U
        }
        io.out := w
      }, "treadle") {
        c => new PeekPokeTester(c) {
	  poke( c.io.en, 1)
	  expect( c.io.out, 7)
	}
      } should be (true)
    }

}
````

closed 2019-01-23 by stevenmburns