chipsalliance/chisel #993: Inconsistent printf with firrtl backend

immesys, 2019-01-21:
**Type of issue**: bug report 

I am seeing mind-boggling printf statement when I simulate with the FIRRTL backend.

The following excerpt:
```
var y = Wire(UInt(10.W))
var regY = RegNext(y)
var regRegY = RegNext(regY)
...
when (regRegY === 480.U) {
   printf("+++ ry=%x isZero=%x is480=%x\n", regRegY, regRegY === 0.U, regRegY === 480.U);
}
```
Prints

```
+++ ry=0 isZero=0 is480=1
```

This doesn't seem to happen with Verilator as a backend.

I don't know what sort of information would be useful here, if someone can let me know what sort of thing could cause this I can try work it out.

chick, 2019-01-22:
I believe this is a dependency ordering problem in the interpreter. I don't think it is a problem with the *treadle* backend. I will add to my list to test and fix.

chick, 2019-01-22:
Contrary to what I said earlier, I can only reproduce this with treadle. (Though it still might be the case that the interpreter can have the problems). I have created [/freechipsproject/treadle/pull/81](/freechipsproject/treadle/pull/81). It includes a test in PrintStopSpec at the end that is my extrapolation of the problem described above

ucbjrl, 2019-11-14:
@chick - can we close this?

chick, 2019-11-15:
I think we can close it. Treadle is the supported scala backend.

closed 2019-11-15 by chick