chipsalliance/chisel #996: Latest merge with experimental fails

tampler, 2019-01-22:
**Type of issue**: bug report 

Two latest commits by Richard Lin on Jan21 break output generation. 

[info] my_pkg.MaccTester *** ABORTED ***
[info]   java.lang.NoClassDefFoundError: chisel3/core/ImplicitModule
[info]   at chisel3.iotesters.setupTreadleBackend$.apply(TreadleBackend.scala:141)
[info]   at chisel3.iotesters.Driver$.$anonfun$execute$2(Driver.scala:54)

This fails to compile, but release from Jan 18 works fine with my code [here](https://github.com/Neurodyne/chisel_katas/blob/master/src/test/scala/VrfTest.scala)

Any support required ?

jackkoenig, 2019-01-22:
This sounds like a binary incompatibility from https://github.com/freechipsproject/chisel3/pull/994. Can you provide more information about your flow? Are you publishing the projects locally yourself or pulling from the published snapshots?

In any case, this looks like your chisel-testers were compiled against a commit pre #994 but the chisel3 getting pulled in is post #994. You just need to make sure your testers are compiled against a newer local publish of chisel3.

If you're using the snapshots that are published on the web, then it sounds like there's an issue with our snapshot publishing.

tampler, 2019-01-22:
Hi Jack

I'm pulling and building FIRRTL, Chisel3 and chisel-testers from master and use local snapshots.
Scala is 2.12.8 and sbt is 1.2.8

I get this error for all commits, starting from Jan21. I rolled back to Jan18 commits and all works seamlessly. I test with [chisel_katas](https://github.com/Neurodyne/chisel_katas/) project

albert-magyar, 2019-01-22:
As Jack says, it sounds like you need to re-run `publishLocal` in `chisel-testers to get one that is compiled against latest chisel3. 

However, the unfortunate truth is that it would be reasonable to try deleting your project classfiles to force the issue. I know I am not the only person to run into similar situations where sbt occasionally chokes.

tampler, 2019-01-22:
same for commit eb6ddf4e0da1d3fa9e506e0b5f26ff87c411665e on Jan 22 by Schuyler

tampler, 2019-01-22:
@albert-magyar Removed all binary files from chisel-testers and rebuilt it from scratch. Didn't help as well. Something is wrong here

ucbjrl, 2019-01-22:
@ducky64's PR renames `ImplicitModule` to `MultiIOModule` which explains why you'll get a:
```
[info] my_pkg.MaccTester *** ABORTED ***
[info]   java.lang.NoClassDefFoundError: chisel3/core/ImplicitModule
```
What's curious is why the stack trace claims:
```
[error] 	at chisel3.iotesters.setupTreadleBackend$.apply(TreadleBackend.scala:141)
```
It looks like when treadle support was added six months ago, the Module type parameter was:
```scala
  def apply[T <: MultiIOModule](
```
Make sure your code `MaccTest.scala:21` isn't specifying something that generates an `ImplicitModule` as the `dut` argument to `Driver.execute`.

albert-magyar, 2019-01-22:
I tried it out, and I am able to run `testOnly my_pkg.MaccTester -- -z Basic` using master on all the repos (firrtl, chisel3, chisel-testers, chisel_katas).

tampler, 2019-01-22:
@albert-magyar Which OS are you running ? Hmm that's strange. I'll try that again tomorrow from scratch

albert-magyar, 2019-01-22:
OS X 10.13.6

ducky64, 2019-01-22:
I've had errors with sbt preferring sonatype versions over local versions even if local versions were published more recently. I'm not sure how to fix this, but this might be your problem, especially since your log says
```
[info] Updating ...
[warn] Resolving a snapshot version. It's going to be slow unless you use `updateOptions := updateOptions.value.withLatestSnapshots(false)` options.
[info] Out of 1 candidates we found for edu.berkeley.cs#chisel3_2.12;3.2-SNAPSHOT in sonatype-snapshots, we are choosing sonatype-snapshots.
[warn] Sorting results from edu.berkeley.cs#firrtl_2.12;1.2-SNAPSHOT, using Tue Jan 22 22:13:15 MSK 2019 and Tue Jan 22 22:13:15 MSK 2019.
[warn] Sorting results from edu.berkeley.cs#firrtl_2.12;1.2-SNAPSHOT, using Tue Jan 22 22:13:15 MSK 2019 and Tue Jan 22 22:13:15 MSK 2019.
[warn] Resolving a snapshot version. It's going to be slow unless you use `updateOptions := updateOptions.value.withLatestSnapshots(false)` options.
[info] Out of 2 candidates we found for edu.berkeley.cs#firrtl_2.12;1.2-SNAPSHOT in local and sonatype-snapshots, we are choosing local.
[info] Out of 2 candidates we found for edu.berkeley.cs#firrtl_2.12;1.2-SNAPSHOT in local and sonatype-snapshots, we are choosing sonatype-snapshots.
[warn] Resolving a snapshot version. It's going to be slow unless you use `updateOptions := updateOptions.value.withLatestSnapshots(false)` options.
[info] Out of 1 candidates we found for edu.berkeley.cs#chisel-iotesters_2.12;1.3-SNAPSHOT in sonatype-snapshots, we are choosing sonatype-snapshots.
[warn] Resolving a snapshot version. It's going to be slow unless you use `updateOptions := updateOptions.value.withLatestSnapshots(false)` options.
[info] Out of 1 candidates we found for edu.berkeley.cs#firrtl-interpreter_2.12;1.2-SNAPSHOT in sonatype-snapshots, we are choosing sonatype-snapshots.
[warn] Resolving a snapshot version. It's going to be slow unless you use `updateOptions := updateOptions.value.withLatestSnapshots(false)` options.
[info] Out of 1 candidates we found for edu.berkeley.cs#treadle_2.12;1.1-SNAPSHOT in sonatype-snapshots, we are choosing sonatype-snapshots.
[info] Done updating.
```

ucbjrl, 2019-01-22:
Unfortunately, the sbt diagnostic output is incorrect for at least two reasons:
 - for each candidate, it claims to be choosing that candidate ([check out the sbt source code](https://github.com/sbt/librarymanagement/blob/bb2c73e183fa52e2fb4b9ae7aca55799f3ff6624/ivy/src/main/scala/sbt/internal/librarymanagement/ivyint/SbtChainResolver.scala)),
 - it doesn't understand that after a successful `publishLocal` (at lease on MacOS and Unix), what's in the `.ivy2/cache/...` directory isn't the actual Jar file, but a reference to the Jar file in the `.ivy2/local/... directory.

tampler, 2019-01-22:
@ucbjrl Jim, Can I rebuilt treadle separately ? I didn't see a separate repo for it

If not possible, I'll clean cache and local in my ~/.ivy2 and will try from scratch.

ucbjrl, 2019-01-22:
Yes. The treadle repo is https://github.com/freechipsproject/treadle.

ucbjrl, 2019-01-23:
freechipsproject/chisel3#998 has added aliases for `UserModule` and `ImplicitModule` so this should be resolved with a new pull. Still, it would be instructive to know why this was failing - who was referring to `ImplicitModule`?

jackkoenig, 2019-01-23:
@ucbjrl, even if no one directly refers to `ImplicitModule`, isn't there a binary incompatibility between projects compiled against chisel3 where `UserModule` is an alias for `ImplicitModule` and those where `ImplicitModule` is renamed? I'm actually unsure if #998 fixes that problem.

ucbjrl, 2019-01-23:
Possibly, but we're talking about SNAPSHOT (i.e., bleeding edge) versions, and I don't believe we currently make any guarantee about binary compatibility between SNAPSHOT releases.
We should discuss this at the next dev meeting.

tampler, 2019-01-24:
Thanks everybody for feedback

closed 2019-01-24 by tampler