chipsalliance/chisel #1012: Vec of modules

coogsie, 2019-02-12:
**Type of issue**: bug report | feature request | other enhancement

**Impact**: no functional change | API addition (no impact on existing code) | API modification | unknown

**Development Phase**: request | proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

edwardcwang, 2019-02-12:
Hi there, and thank you for your interest in Chisel!

A `Vec` in Chisel is a type (like `UInt`, `Bool`, etc). For this reason, we can only have `Vec of <type>` not `Vec of <piece of hardware e.g. Mux, Module, etc>`. It is the same reason we can't have "Vec of Registers" in Chisel: https://github.com/freechipsproject/chisel3/wiki/Cookbook#how-do-i-create-a-vector-of-registers

coogsie, 2019-02-12:
Hi, the way you are using is different to that of the UCB Chisel wiki (https://github.com/ucb-bar/chisel-tutorial/wiki/instantiating-modules)

`val foo = Vec(Seq.fill(16){ Module(new bar( i ) ).io } )`
vs
`val foo = (0 until 10).map { i => Module(new bar( i ) ) }`

If the way you specified is the correct way then I apologize, I am still learning the nuances of Chisel and Scala.

edwardcwang, 2019-02-12:
So the former creates a `Vec of <bar io>`, where `<bar io>` is (typically) a Bundle defining the IOs of the module `bar`, whereas the latter creates a Scala `Seq` of Modules. You can transform the latter into the former via `Vec(foo.map(_.io).toSeq)`. It's just that the `Vec` itself can't be  dependent upon arbitrary hardware.

edwardcwang, 2019-04-11:
Please feel free to re-open or leave a comment if you have any further questions.

closed 2019-04-11 by edwardcwang