chipsalliance/chisel #1020: RegNext logic for Bundle should be updated

tampler, 2019-02-16:
I'm giving a self explanatory code below. My desire is to update the **RegNext** implementation, so that I don't need to DRY and write a double bulk connect. Perhaps, that's about the BiConnect logic, but I can't currently fix that on my own. Using FIRRTL and Chisel3 Snapshots

```scala
class DspShell extends Module {
  val io = IO(new Bundle {
    val vld = Input(Bool())
    val clr = Input(Bool())
    val a   = Input(inType)
    val b   = Input(inType)
    val rdy = Output(Bool())
    val res = Output(outType)
  })
  val dut  = Module (new DspTop)  
  dut.io <> RegNext(io) // <------ this should do the job. Don't need to write two bulk connects like this
  io <> RegNext(dut.io)
}

```

albert-magyar, 2019-02-22:
I think changing assignment semantics to support this with registers is probably not the right direction, given how deep the changes to the language would be.

With current Chisel semantics in place, it would be possible to define a utility `BidirectionalPipe`. As long as it's a *module*, and not a register, it wouldn't require any changes to assignment semantics. 

```
object BidirectionalPipe {
  def apply[T <: Data](bidir: T) = {
    val pipe = new BidirectionalPipe(bidir) // type-parameterized pipe wrapper
    TransitName(pipe.io, pipe)
  }
}
```

Then, you would be able to write `dut.io <> BidirectionalPipe(io)` with no changes to the existing language features.

schoeberl, 2019-02-23:
For a clear pipeline design, I would split the input and output signals into subbundles. And then it can be made explicit where the registers are placed. That both directions are registered at the same place/module looks for me more an antipattern than good design.

I personally would like that the code where a register input is fed by an input signal should result in an error. In most cases, this is probably a design error.

tampler, 2019-03-23:
Thank you for your feedback. That's not a critical feature, so we don't need to redesign a lot just to make this happen. Closing the issue

closed 2019-03-23 by tampler