chipsalliance/chisel #1029: Mismatch between wiki saying bitwise reductions work on SInts and implementation which does not

jackkoenig, 2019-02-28:
h/t @redpanda3 for noticing the mismatch

**Type of issue**: bug report | documentation

**Impact**: no functional change | API addition (no impact on existing code)

**Development Phase**: request

**Other information**
 

**What is the current behavior?**

Bitwise reduction operators are only defined on `UInt`, **not** on `SInt`

**What is the expected behavior?**

This Chisel wiki says they're defined on both `UInt` and `SInt`:
https://github.com/freechipsproject/chisel3/wiki/Builtin-Operators

**What is the use case for changing the behavior?**

I'm not sure if the behavior should change or if the documentation should change. I suspect the operators are only defined on `UInt` for a reason, and while my intuition is that `andR` and `orR` are fine on `SInt`s, `xorR` seems like it could interact in surprising ways with width inference.

I just wanted to make sure we fix the mismatch.

edwardcwang, 2019-02-28:
I would vote we fix the documentation to say they're defined only on UInt (really being used as `Bits`), since semantically it's not clear you'd want to bitwise reduce on a signed integer.

ekiwi, 2019-05-08:
In firrtl they are defined on both, SInt and UInt:

[image]

ekiwi, 2019-05-21:
@jackkoenig : should we remove the `SInt ` Arg Type from the Firrtl spec?

jackkoenig, 2019-05-21:
More of a question for @azidar on the Firrtl spec.

Note that FIRRTL and Chisel don't have to be in sync on this, Chisel can define them only on `UInt` while FIRRTL can continue to support both if it wants