chipsalliance/chisel #1030: Generic Type for Module Generation

zqxkb7fh, 2019-03-05:
It is allowed to use "Parameterized Classes" for generate different types of IO for modules:
https://github.com/freechipsproject/chisel3/wiki/Polymorphism-and-Parameterization

But what if I want to make instances of Module depends on the generic type? For example,

class X[T] extends Module {
    ...
    val subMod = Module(new T)
    ...
}

It seems to yield a error message that class type required after the keyword "new". 
Is it possible to achieve that? Thanks.

edwardcwang, 2019-03-05:
In general, it's not possible for the compiler to obtain the constructor for `T` via `T` alone. We have to provide a parameter `genT: => T` which is a function that returns a new T (effectively, calling `genT` is equivalent to `new T`).

Note that you will need some additional type information to access any IOs of the module you are instantiating. See below for a complete example.

```scala
import chisel3.experimental.{BaseModule, RawModule}

// Provides a more specific interface since generic Module
// provides no compile-time information on generic module's IOs.
trait MyAdder {
    def in1: UInt
    def in2: UInt
    def out: UInt
}

class Mod1 extends RawModule with MyAdder {
    val in1 = IO(Input(UInt(8.W)))
    val in2 = IO(Input(UInt(8.W)))
    val out = IO(Output(UInt(8.W)))
    out := in1 + in2
}

class Mod2 extends RawModule with MyAdder {
    val in1 = IO(Input(UInt(8.W)))
    val in2 = IO(Input(UInt(8.W)))
    val out = IO(Output(UInt(8.W)))
    out := in1 - in2
}

class X[T <: BaseModule with MyAdder](genT: => T) extends Module {
    val io = IO(new Bundle {
        val in1 = Input(UInt(8.W))
        val in2 = Input(UInt(8.W))
        val out = Output(UInt(8.W))
    })
    val subMod = Module(genT)
    io.out := subMod.out
    subMod.in1 := io.in1
    subMod.in2 := io.in2
}

println(chisel3.Driver.emitVerilog(new X(new Mod1)))
println(chisel3.Driver.emitVerilog(new X(new Mod2)))
```

zqxkb7fh, 2019-03-06:
Thank you!
And one thing to notice is that the _desiredName_ method should be override defined to distinguish modules with different submodule instanced.

edwardcwang, 2019-03-06:
Great! Please feel free to open another issue if you have any further questions about Chisel.

closed 2019-03-06 by edwardcwang

Jagpreet-Sifive, 2019-08-12:
can you  explain  (<: )  symbol uses in parameterization ?

edwardcwang, 2019-08-12:
`<:` means an "upper type bound" in Scala. That is, if you have a `class MyClass[T <: Plant]`,  the type parameter `T` must be an instance of `Plant` or a subclass of `Plant`.

For more information: https://docs.scala-lang.org/tour/upper-type-bounds.html