chipsalliance/chisel #1039: The contradiction of Width Inference

kai413629305, 2019-03-18:
In Width Inference, the result of & primitive operation is the maximum or minimum of input A and B bit-width. Why is the minimum value of two inputs stated in the wike of Chisel3 and the maximum value in the source code of Chisel3? Which is right?
chisel3 wiki: https://github.com/freechipsproject/chisel3/wiki/Width-Inference
chisel3 source code:chisel3/chiselFrontend/src/main/scala/chisel3/core/Bits.scala:Line 785

jackkoenig, 2019-03-18:
Good catch, it is in fact the maximum width. Chisel (well FIRRTL) will sign extend the other input and then and them together. I've fixed the wiki. Thanks!

closed 2019-03-18 by jackkoenig

kai413629305, 2019-03-19:
> Good catch, it is in fact the maximum width. Chisel (well FIRRTL) will sign extend the other input and then and them together. I've fixed the wiki. Thanks!

Why is  the maximum width? eg: 10101010 & 1110 should be equal to 1010, according to the maximum width is 00001010.

jackkoenig, 2019-03-20:
While that's true for UInts it's not true for SInts where negative numbers will sign extend with a 1. I'm not sure why max width was chosen, but it makes it consistent with all of the other bitwise logical operators rather than `&` of `UInts` being a special case.