chipsalliance/chisel #1042: Register declared inside a when block executes unconditionally

furkanusta, 2019-03-19:
**Type of issue**: bug report

**Impact**: unknown (possibly no functional change)

**Development Phase**: request

**Other information**

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

```Scala
class Foo extends Module {
  val io = IO(new Bundle {
    val start  = Input(Bool())
    val output = Output(Bool())
  })
  io.output := false.B
  when(io.start) {
    val counter = RegInit(0.U(16.W))
    counter := counter + 1.U
    io.output := counter === 15.U
  }
}
```

**What is the current behavior?**
counter increments at each cycle

**What is the expected behavior?**
counter should increment only when io.start is high, or if possible a warning can be shown.

**Please tell us about your environment:**
    - version: `3.2 (trunk 2c449c5d6e23dcbb60e8c64cab6b6f4ba6ae313f)
    - OS: Centos 7

**What is the use case for changing the behavior?**
This is not exactly a request for a change since the fix is trivial for the above code and I don't think it is a common coding style but I wanted to clarify whether this was the intended behavior. 

This is what I believe to be the Verilog equivalent of the above Chisel code and it works as I expected (though it requires the blocks to be named)
```Verilog
module Foo(input start, clock, output reg out);
    always @(posedge clock) begin : block
        reg[4:0] counter = 5'b0;
        out <= 0;
        if (start) begin
            counter <= counter + 1;
            out <= counter == 15;
        end
    end
endmodule
```

aswaterman, 2019-03-19:
Yes, this is the expected behavior. The update condition is derived from the conjunction of the when-statements between the declaration and the assignment. Since they’re declared in the same scope as each other, there are no such when-statements, so it’s an unconditional assignment.

furkanusta, 2019-03-19:
Thanks for the quick response. I was expecting it to be intended but wanted to be sure

closed 2019-03-19 by furkanusta