chipsalliance/chisel #1058: Expand APIs that support ChiselEnum

jackkoenig, 2019-03-30:
**Type of issue**: feature request

**Impact**: API addition (no impact on existing code)

**Development Phase**: request

**Other information**
 

Inspired by this question on StackOverflow: https://stackoverflow.com/questions/55432650/how-to-use-a-chisel3-experimental-chiselenum-in-an-i-o-port

**What is the current behavior?**

`MuxLookup` only accepts `UInts` as selector

**What is the ~expected~ desired behavior?**

Things like `MuxLookup` should/could support `ChiselEnum`. I'm sure there are other things that should accept `ChiselEnum`s as well.

**What is the use case for changing the behavior?**

More convenient than having to cast every enum to UInt.

ducky64, 2019-03-31:
A general solution might be related to the type system discussion in #1056 - does ChiselEnum provide stronger guarantees than Element? Would it make sense for it to be a Bits subtype instead (might also need a resolution to #957), and utility functions that generate logic should be able to operate on Bits instead of UInt?

jackkoenig, 2021-02-17:
This has come up again, this time with `switch`: https://stackoverflow.com/questions/66165591/chisel-passing-enum-type-as-io

I know it's been almost 2 years but I'm going to answer @ducky64's original questions

> does ChiselEnum provide stronger guarantees than Element?

Yes, but so does every subclass of Element.

> Would it make sense for it to be a Bits subtype instead (might also need a resolution to #957), and utility functions that generate logic should be able to operate on Bits instead of UInt?

For some, maybe, but not for others. While it makes sense for `switch` and `MuxLookup` to key off of both `UInt` and `ChiselEnum`, I don't think it makes sense for them to be keyed off of `SInt` or `FixedPoint`. I think in this case it just makes sense to overload the `MuxLookup` apply method.

jackkoenig, 2021-02-17:
`switch` actually [already accepts subtypes of `Element`](https://www.chisel-lang.org/api/latest/chisel3/util/switch$.html#apply[T%3C:chisel3.Element](cond:T)(x:=%3EAny):Unit) which probably makes sense, perhaps we should just do the same for `MuxLookup` and others.