chipsalliance/chisel #1063: Weird binding/directionality issue with Vec(n, Flipped(...))

edwardcwang, 2019-04-11:
**Type of issue**: bug report

**Development Phase**: request 

**Other information**: Chisel 3.2, revision 9f77bf54543724594711f0455ef91f51edfbd22e

Consider the following snippet of code:
```scala
import chisel3._
import chisel3.experimental.{DataMirror, RawModule}
class MyBundle extends Bundle {
    val a = Input(Bool())
    val b = Output(Bool())
}

class MyModule(val n: Int) extends RawModule {
  val proto = new MyBundle
  // def proto = new MyBundle <-- same problem

  val foo = IO(proto)
  val vecFoo = IO(Vec(n, proto))

  val bar = IO(Flipped(proto))
  val vecBar = IO(Vec(n, Flipped(proto)))

  foo <> DontCare
  vecFoo <> DontCare
  bar <> DontCare
  vecBar <> DontCare

  // directionOf prints the wrong direction deceptively
  println(DataMirror.directionOf(vecBar.head.a)) // returns Output
  println(DataMirror.directionOf(vecBar.head.b)) // returns Input
}
```

The `vecBar` line generates incorrect Verilog:

```verilog
module MyModule(
  input   foo_a,
  output  foo_b,
  input   vecFoo_0_a,
  output  vecFoo_0_b,
  input   vecFoo_1_a,
  output  vecFoo_1_b,

  // IO(Flipped()) works fine
  output  bar_a,
  input   bar_b,

  // But Vec(n, Flipped()) generates wrong directions
  input   vecBar_0_a, // <-- this should be output!
  output  vecBar_0_b, // <-- this should be input!
  input   vecBar_1_a,
  output  vecBar_1_b
);
  assign foo_b = 1'h0;
  assign vecFoo_0_b = 1'h0;
  assign vecFoo_1_b = 1'h0;
  assign bar_a = 1'h0;
  assign vecBar_0_b = 1'h0;
  assign vecBar_1_b = 1'h0;
endmodule
```

edwardcwang, 2019-04-11:
This could be a FIRRTL bug actually

```
circuit MyModule :
  module MyModule :
    output foo : {flip a : UInt<1>, b : UInt<1>}
    output vecFoo : {flip a : UInt<1>, b : UInt<1>}[2]
    input bar : {flip a : UInt<1>, b : UInt<1>}
    output vecBar : {flip a : UInt<1>, b : UInt<1>}[2]
    ; ...
```

jackkoenig, 2019-04-11:
Weird, this looks very similar to https://github.com/freechipsproject/chisel3/pull/945, but the tests I added there didn't actually invoke FIRRTL (although I did try #945 on designs and it fixed the problem there...)

edwardcwang, 2019-04-11:
Yep, upon further testing this is a bug in FIRRTL. PR forthcoming.

edwardcwang, 2019-04-11:
Simplest reproducing code:
```scala
import chisel3._
import chisel3.experimental.{DataMirror, RawModule}
class MyBundle extends Bundle {
  val a = Input(Bool())
  val b = Output(Bool())
}
class MyModule extends RawModule {
  val vecBar = IO(Vec(1, Flipped(new MyBundle)))

  vecBar <> DontCare

  // directionOf prints the wrong direction deceptively
  println(DataMirror.directionOf(vecBar.head.a)) // returns Output
  println(DataMirror.directionOf(vecBar.head.b)) // returns Input
}

chisel3.Driver.emitVerilog(new MyModule)
```

jackkoenig, 2019-04-11:
Guess I failed in #945 ☹️

ducky64, 2019-04-12:
This could be a problem in the [Vec emitter code](https://github.com/freechipsproject/chisel3/blob/master/src/main/scala/chisel3/internal/firrtl/Emitter.scala#L36) - note how the `Record` case checks into the element's SpecifiedDirection and generates flips as necessary, but there is nothing similar in the `Vec` case. I also wouldn't be surprised if `Vec(Input(...))` wasn't generating flips either.

jackkoenig, 2019-04-12:
I think `emitVerilog` actually uses the `Converter`, although that was more-or-less transliterated from the Emitter so seems plausible the bug would be replecated

commit 2425e940 (2019-04-15): Fix wrong directionality for Vec(Flipped())

Create Chisel IR Port() in a way that Converter is happy with.
Also add more extensive test suite for future-proofing.

Close #1063
commit 8774a3e3 (2019-04-15): Fix wrong directionality for Vec(Flipped())

Create Chisel IR Port() in a way that Converter is happy with.
Also add more extensive test suite for future-proofing.

Close #1063
commit 967bb73d (2019-04-19): Fix wrong directionality for Vec(Flipped())

Create Chisel IR Port() in a way that Converter is happy with.
Also add more extensive test suite for future-proofing.

Close #1063
PR #1064 (2019-04-19): Fix wrong directionality for Vec(Flipped())
Create Chisel IR Port() in a way that Converter is happy with.
 Also add more extensive test suite for future-proofing.
 
Close #1063
closed 2019-04-19 by edwardcwang