chipsalliance/chisel #1072: Make wrapper class like DecoupledIO as functor

haedaal, 2019-04-18:
**Type of issue**: feature request

**Impact**: API addition (no impact on existing code)

**Development Phase**: request | proposal

**Other information**
Bit tired of writing boiler plate to connect DecoupledIO for every tiny job, I started to feel it neccessary to have 'map' on DecoupledIO
What's just working is
```scala
object ChiselFunctor {
  implicit class DecoupledFuntor[T1 <: Bundle](d: DecoupledIO[T1]) {
    def map[T2 <: Data](f: T1 => T2) : DecoupledIO[T2] = {
      val new_bits = f(d.bits)
      val new_decoupled = Wire(Decoupled(chiselTypeOf(new_bits)))
      new_decoupled.ready := d.ready
      new_decoupled.valid := d.valid
      new_decoupled.bits := new_bits
      new_decoupled
    }
  }
}
```
And after writing, I thought It would be great if this kind of functionality can be integrated, managed, and supported officially.

thanks for reading!

ducky64, 2019-04-19:
Resolution from today's meeting is to add this kind of API. Specific implementation and how the API will be exposed TBD, but it will probably look like what you have, just maybe in the base DecoupledIO class instead of needing a separate import + implicit conversion.

ducky64, 2019-04-19:
Other things to think about: how much to scale this, eg should we also add ValidIO.map

jackkoenig, 2022-08-01:
Fixed by https://github.com/chipsalliance/chisel3/pull/2646, released in v3.5.4.

closed 2022-08-01 by jackkoenig