chipsalliance/chisel #1090: Allow internal tests to set whether or not a VCD should be dumped

edwardcwang, 2019-05-07:
**Type of issue**: other enhancement

**Impact**: no impact to user code

**Development Phase**: request

**Other information**: Some long running tests (as noticed in #1088, though exhaustive testing may not be the right approach in a unit test) may want to be able to control or suppress if a VCD is generated.

Chisel's `TestDriver` calls FIRRTL's `verilogToCpp` which has VCD generation turned on by default.

https://github.com/freechipsproject/chisel3/blob/7da0d5f8704b25a94e25cd9638c77695ae081d7f/src/main/scala/chisel3/testers/TesterDriver.scala#L60

https://github.com/freechipsproject/firrtl/blob/a7cf6ff3416a11088d811a435ba71fd36b191fb4/src/main/scala/firrtl/util/BackendCompilationUtilities.scala#L101

seldridge, 2019-05-07:
My specific annoyance is that `assertTesterPasses` is generating VCDs by default. Repeated runs of `sbt test` keep filling up `test_run_dir` with a lot of VCD files that aren't being checked.

jackkoenig, 2019-11-18:
Turn off VCD generation by default, add flag to turn them on? It's also probably reasonable for developers to just change the Scala to turn it back on.