chipsalliance/chisel #1097: strange problem when i use when and otherwise

minisparrow, 2019-05-15:
strange problem happened that when condition is fulfilled, expression 1 executed but expression 2 also executed. 
```bash
when(condition) {
   expression 1
} .otherwise{
   expression 2
}
```

ccelio, 2019-05-15:
You'll need to show us a lot more code than that to get an idea of what the issue could possibly be.

Are you by chance using `var` variables in combination with the scala `=` assignment inside of your `when`/`otherwise` statements?

seldridge, 2019-05-15:
A full example showing what you are doing may help, but I'm guessing that you may be getting confused between elaboration and the actual generated hardware as well as what is a Chisel type and what is a Scala type. Taking a stab, but possibly in the wrong direction.

Every statement inside a Chisel module, that would follow the normal execution of a Scala program is executed. The internals of a Chisel module are the description of what you want your hardware to look like. In you specific example you're describing a mux keyed by a condition. `when` and `.otherwise` are parts of Chisel. These are not the same as `if` and `else` which are Scala constructs.

Consider the following simple module. This mixes `println` (a Scala print) inside `when`/`otherwise`. During elaboration, both `println`s print, but you wind up getting the correct hardware. 

```scala
class FooIO extends Bundle {
  val a: Bool = Input(Bool())
  val b: Bool = Output(Bool())
}

class Foo extends Module {

  val io: FooIO = IO(new FooIO)

  val isTrue: Bool = io.a === true.B
  val notA: Bool = ~io.a

  when (isTrue) {
    println("[info] elaborate true condition")
    io.b := io.a
  } .otherwise {
    println("[info] elaborate otherwise condition")
    io.b := notA
  }

}
```

This then produces the following log text:
```
[info] [0.003] Elaborating design...
[info] elaborate true condition
[info] elaborate otherwise condition
[info] [0.120] Done elaborating.
Total FIRRTL Compile Time: 903.5 ms
```

And the following Verilog:
```verilog
module Foo(
  input   clock,
  input   reset,
  input   io_a,
  output  io_b
);
  wire  notA;
  assign notA = ~ io_a; 
  assign io_b = io_a ? io_a : notA;
endmodule
```

There's a [similar StackOverflow question/answer](https://stackoverflow.com/questions/51644857/when-we-should-use-not-in-chisel3-same-case-is-when-and-if/) that may help shine some light on this.

minisparrow, 2019-05-15:
@ccelio @seldridge Thanks for reply. The code is here. I want to assign inputs_pad when row or col is not zero, in which case the index of io.inputs can not be out of bounds. But when elaborating, it will execute this assignment "inputs_pad(idx) := io.inputs(idx_in)",but when row = 0 ,the idx_in is negative. so it gives the error message 
```bash
java.lang.IndexOutOfBoundsException: -6
```
code is here
```bash
  val in_pad_width = 2*pad + width
  val in_pad_height = 2*pad + height
  val inputs_pad = Reg(Vec(in_pad_width * in_pad_height, UInt(w.W)))

  for (row <- 0 until in_pad_height){
     for (col <- 0  until in_pad_width){
         val idx = row * in_pad_width
         when(row.asUInt(w.W) === 0.U || col.asUInt(w.W) === 0.U ){
             inputs_pad(idx) := 0.U
             //println(s"true, row = $row, col = $col ")
         }.otherwise{
             val idx_in = (row-1) * width + col - 1 
             inputs_pad(idx) := io.inputs(idx_in)
             //println(s"false, idx_in = $idx_in, row = $row, col = $col ")
         }
     }
  }
```

chick, 2019-05-16:
Here's my best guess at what you are trying to do, in particular I am guessing at the structure of your inputs. This should still provides some help even if I got this assumption wrong
```scala
class PadProblem(pad: Int, width: Int, height: Int, w: Int) extends Module {
  val in_pad_width  = 2*pad + width
  val in_pad_height = 2*pad + height
  val padSize = in_pad_width * in_pad_height

  val io = IO(new Bundle {
    val inputs = Vec(width * height, Input(UInt(w.W)))
  })

  val inputs_pad    = Seq.fill(in_pad_width * in_pad_height){ Reg(UInt(w.W)) }

  for (row <- 0 until in_pad_height) {
    for (col <- 0  until in_pad_width) {
      val idx = (row * in_pad_width) + col

      if(row == 0 || col == 0 || row == in_pad_height - 1 || col == in_pad_width - 1) {
        println(s"padding at ($row,$col) idx $idx")
        inputs_pad(idx) := 0.U
      }
      else {
        val idx_in = ((row-1) * width) + (col - 1)
        println(s"connecting ($row, $col) $idx from $idx_in")
        inputs_pad(idx) := io.inputs(idx_in)
      }
    }
  }
}
```

Notes
- It seems like you creating a 2d set of registers that are mapped to a 2d set of inputs
- The set of registers are padded around the outside
  - The pad registers are initialized to `0.U`
- Unless you need to use hardware addressing of a structure you do not need to create a Vec
  - Instead just keep track of them using Scala structures
- You do not need `when`, instead you need to conditionally construct hardware
  - That means use scala `if`
  - println helps show how it initializes during construction
- The above elaborates correctly given my assumptions
- Let me if this helps

chick, 2019-05-16:
This is an extra refinement of the code above. Assuming in fact what is going on here is some 2 dimensional mapping of inputs into a register array, you should take advantage of the ability to represent the problem in a 2D fashion in your code. I think it makes this easier to read and easier to reason about. Here is an example of the same Module but written in 2D model.
```scala
class PadProblem2D(pad: Int, width: Int, height: Int, w: Int) extends Module {
  val in_pad_width  = 2*pad + width
  val in_pad_height = 2*pad + height

  val io = IO(new Bundle {
    val inputs = Vec(height, Vec(width, Input(UInt(w.W))))
  })

  val inputs_pad    = Seq.fill(in_pad_width, in_pad_height){ Reg(UInt(w.W)) }

  // put zeros around edges
  // first do bottom and top row
  inputs_pad.head.foreach { cell => cell := 0.U }
  inputs_pad.last.foreach { cell => cell := 0.U }
  // do first and last column
  inputs_pad.indices.foreach { index =>
    inputs_pad(index).head := 0.U
    inputs_pad(index).last := 0.U
  }

  // wire the interior of the padded registers to the inputs
  for (row <- 0 until height) {
    for (col <- 0  until width) {
      inputs_pad(row + 1)(col + 1) := io.inputs(row)(col)
    }
  }
}
```
Note:
- I have made the inputs a Vec of Vecs (2D)
- I have used Seq.fill in a 2D context to create the register array
- I initialize the outside of the register array using scala seq methods
- The interior of the register array is more neatly described using 2d indexing
- The generated firrtl is not any shorter but the final names are more readable as 2D indices
- You could also rely on Chisel's last connect semantics to initialize them all to zero
  - your later connects would override the zeros in the interior of the register array
  - note the RegInit instead of Reg
  - you could then remove the `//put zeroes around the edges`` code section
```scala
  val inputs_pad    = Seq.fill(in_pad_width, in_pad_height){ RegInit(0.U(w.W)) }
```

chick, 2019-05-16:
One further thought. This is a good question, but as a question (as opposed to an issue or problem with Chisel) It would probably be better to ask it on [StackOverflow](https://stackoverflow.com/) and tag it with the #chisel tag. For questions stackoverflow makes the answers a bit easier to find for other users experiencing similar questions. We regularly monitor stackoverflow and try to answer promptly
-cheers

minisparrow, 2019-05-18:
@chick Thanks very much for your help. Great help for me. and I also found my error in my code. Thank you very very very much :)

minisparrow, 2019-05-18:
> One further thought. This is a good question, but as a question (as opposed to an issue or problem with Chisel) It would probably be better to ask it on [StackOverflow](https://stackoverflow.com/) [...]

@chick OK. I will ask question on StackOverflow next time.Thank you very much.

closed 2019-05-18 by minisparrow