chipsalliance/chisel #1100: @chiselName on a Module with a ChiselEnum results an error

kammoh, 2019-05-22:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**
 

Applying `@chiselName` annotation on a `Module` with a `ChiselEnum` object results in compilation error.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

``` scala
@chiselName
class MyModule extends MultiIOModule {
  object State extends ChiselEnum {
    val x, y, z = Value
  }
}
```

Scala compile error:
`Error: Value cannot be called without assigning to an enum
    val x, y, z = Value
`
**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

- version: `3.0-SNAPSHOT`
- OS: `macOS`

**What is the use case for changing the behavior?**

hngenc, 2019-05-23:
It seems the problem is that @chiselName will change `val x = Value` to `val x = context.name(Value, x)`, at least judging by this line: https://github.com/freechipsproject/chisel3/blob/820e2688f0cb966d4528ff074fdbdc623ed8f940/coreMacros/src/main/scala/chisel3/internal/sourceinfo/NamingAnnotations.scala#L156

ducky64, 2019-05-23:
That sounds like a reasonable root-cause - the chiselName macro really mostly is a simple rewrite that adds a `context.name` call to anything of the form `val name = expr`. Blacklisting `Value` would be one possible solution, though it's not a selective / general one. Thoughts? Perhaps if type information is available to macros, we could whitelist only expressions that type to Data?

hngenc, 2019-05-24:
My plan was to change the `Value` macro so that it also recognizes statements of the form `val x = foo(x, ...)`. Macros can be tricky, but I think that would be possible.

PR #1104 (2019-06-24): Fixes @chiselName issue for modules with nested ChiselEnum

**Related issue**: #1100 

**Type of change**: bug fix

**Impact**: no functional change

**Development Phase**: implementation

**Release Notes**

This pull request removes the use of the deprecated `enclosingClass` function in ChiselEnum's `Value` macro. Coincidentally, this also fixes #1100, where modules with a nested ChiselEnum would not work when the `@chiselName` macro was applied to them.
closed 2019-06-24 by azidar