chipsalliance/chisel #1105: adding vec initialisation in chisel3 cheat sheet

Martoni, 2019-06-03:
**Type of issue**: documentation

**Impact**: no functional change

**Development Phase**: request

**Other information**
Is it possible to add Vec initialization in the cheat sheet for chisel 3. I don't know exactly where to find the source of this document (cheat sheet).

For example, here a vec initialization of a register file of 31 UInt registers of 32bits width.

` val regfile = RegInit(VecInit(Seq.fill(31)(0.U(32.W))))`

The cheat sheet I'm using is the version 0.5 [here](https://chisel.eecs.berkeley.edu/doc/chisel-cheatsheet3.pdf).

edwardcwang, 2019-06-03:
@ducky64

Martoni, 2019-06-17:
Is there a public repository for the sources of this cheat sheet ?

edwardcwang, 2019-06-17:
Unfortunately there is not (yet) :-(

ducky64, 2019-06-17:
Actually, we recently open-sourced the cheatsheet: https://github.com/freechipsproject/chisel-cheatsheet

A cleaning pass (including adding `VecInit`) is on my to-do list (Soon™), but we also do accept pull requests.

Martoni, 2019-06-17:
Ho great, thanks. I will follow this repository :)

Martoni, 2019-06-17:
Issue added to right repository [here](https://github.com/freechipsproject/chisel-cheatsheet/issues/1)

closed 2019-06-17 by Martoni