chipsalliance/chisel #1112: DontTouchNotFoundException: Component marked dontTouch (DataCache.cacheStateName) not found!

oharboe, 2019-06-23:
**Type of issue**: bug report

**Impact**: no functional change

**Other information**

I can't include the code that causes this problem, and this isn't something that fails in simple examples. I'm not quite sure how to proceed.

[info]   firrtl.transforms.DontTouchAnnotation$DontTouchNotFoundException: Component marked dontTouch (DataCache.cacheStateName) not found!
[info] It was probably accidentally deleted. Please check that your custom transforms are notresponsible and then file an issue on Github.
[info]   at firrtl.transforms.DontTouchAnnotation$.errorNotFound(OptimizationAnnotations.scala:27)
[info]   at firrtl.transforms.DeadCodeElimination.$anonfun$run$6(DeadCodeElimination.scala:293)
[info]   at firrtl.transforms.DeadCodeElimination.$anonfun$run$6$adapted(DeadCodeElimination.scala:287)
[info]   at scala.collection.immutable.List.foreach(List.scala:392)
[info]   at firrtl.transforms.DeadCodeElimination.run(DeadCodeElimination.scala:287)
[info]   at firrtl.transforms.DeadCodeElimination.execute(DeadCodeElimination.scala:335)
[info]   at firrtl.Transform.$anonfun$runTransform$2(Compiler.scala:257)
[info]   at firrtl.Utils$.time(Utils.scala:183)
[info]   at firrtl.Transform.runTransform(Compiler.scala:257)
[info]   at firrtl.SeqTransformBased.$anonfun$runTransforms$1(Compiler.scala:309)
[info]   ...

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

Exception.

**What is the expected behavior?**

No exception.

**Please tell us about your environment:**

Ubuntu 18

"chisel3" -> "3.1.5",
"chisel-iotesters" -> "1.2.7"

**What is the use case for changing the behavior?**

seldridge, 2019-06-23:
This is indicating that by the time `DeadCodeElimination` ran, there exists a `DontTouchAnnotation` pointing at a component (something in the circuit) that no longer existed.

When transforms run, they generate a rename map which will update annotations based on things that were renamed or deleted. However, if you have a custom transform that manipulates the circuit but doesn't report a rename map, you can have this issue. 

First, check the name of the component that got deleted. Then, try to reason about why this was deleted. Did you add a custom transform? Are you manually writing the annotation and have the name wrong? If it's not related to a custom transform, then it may indicate a FIRRTL bug. If that's the case, trying to narrow it down to a minimum working example would help.

oharboe, 2019-06-23:
I dont have a custom transform. I will give it a go to narrow down to an
example that I can make public.

søn. 23. jun. 2019, 20.02 skrev Schuyler Eldridge <notifications@github.com

oharboe, 2019-06-23:
It turns out that the problem was that I was reusing TesterOptionsManager objects. When I do, I'll share some state between tests(in the test_run_dir?). This seems to have caused the error, though I have no idea why.

When I create a fresh TesterOptionsManager object for each test, it works.

chick, 2019-06-23:
Good catch. There is mutable state in the TesterOptionsManager and that is likely the problem. The work @seldridge has done with [Stage/Phases](https://github.com/freechipsproject/firrtl/pull/920) and related PR's covers this and other issues with the current Driver framework.

oharboe, 2019-06-24:
Is the mutable state in TesterOptionsManager by design?

It seems like a trip-wire from a users point of view. I would not have expected TesterOptionsManager objects to be changed by executing the tests, as I got the impression that it's an argument to the tests, not something that the tests should operate on. 

To me how I launch tests is boilerplate code and I'm happy to switch to a more modern version of it.

Is TesterOptionsManager now obsolete?

Is there an example of the new way to launch tests and avoid the old "Driver" method?