chipsalliance/chisel #1114: Stage/Phase exception processing hides expected exceptions

ucbjrl, 2019-06-24:
**Type of issue**: bug report

**Impact**: API modification

**Development Phase**: proposal

**Other information**
`ChiselGeneratorAnnotation.elaborate(gen)` catches exceptions generated in `Builder.build(Module(gen()))` and throws an OptionsException.

This breaks code that's expecting a more specific exception, notably:
```scala
        "UInt subtract with overflow type Grow not supported" in {
          intercept[DspException] {
            dsptools.Driver.execute(() => new BadUIntSubtractWithGrow2(u(4))) { c =>
              new NumbersEmptyTester(c)
            } should be(true)
          }
        }
```
in `dsptools/src/test/scala/dsptools/numbers/NumbersSpec.scala` which now fails with:
```
 ...
[info]     - UInt subtract with overflow type Grow not supported *** FAILED ***
[info]       Expected exception dsptools.DspException to be thrown, but firrtl.options.OptionsException was thrown (NumbersSpec.scala:48)
 ...
[info] *** 1 TEST FAILED ***
[error] Failed: Total 172, Failed 1, Errors 0, Passed 171, Ignored 3
[error] Failed tests:
[error] 	dsptools.numbers.NumbersSpec
[error] (Test / test) sbt.TestsFailedException: Tests unsuccessful

```

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
```bash
$ git clone https://github.com/ucb-bar/dsptools.git
$ cd dsptools
$ sbt clean test
```

**What is the current behavior?**
dsptools.numbers.NumbersSpec fails the `UInt subtract with overflow type Grow not supported` test when a specific `DspException` generates an OptionsException.

**What is the expected behavior?**
`DspException` should be thrown.

**What is the use case for changing the behavior?**
A library may wish to return more specific exceptions to its clients.

seldridge, 2019-06-24:
Is the OptionsException wrapping the DSP exception? If so, should DSP tools be unpacking that or should the pure exception be propagated?

ucbjrl, 2019-06-25:
I think @seldridge's suggestion of unpacking is the expedient solution here. I'll submit a dsptools PR which once accepted, will close this issue.

I'll open another issue for exception/error handling that will propose a more formal API.

jackkoenig, 2019-07-01:
Fixed by https://github.com/ucb-bar/dsptools/pull/168

ucbjrl, 2020-03-21:
Commit d67c4100bd9e24cb9337ad932ba342fd5c49dee4 of PR #1374 revives this issue with the conversion of `OptionsException` to `ChiselException` and the absorption/conversion of `ChiselException` into `StageError` at https://github.com/freechipsproject/chisel3/blob/b8a8d792a0e06520eb1f2acf94e149425234bd3b/src/main/scala/chisel3/stage/ChiselStage.scala#L47

As a result of these changes, the test now fails with:
```
[info] [0.002] Elaborating design...
[error] dsptools.DspException: OverflowType Grow is not supported for UInt subtraction
[error] 	...
[error] 	at dsptools.numbers.UIntRing.minusContext(UIntTypeClass.scala:31)
[error] 	at dsptools.numbers.UIntRing.minusContext$(UIntTypeClass.scala:29)
[error] 	at dsptools.numbers.UIntImpl$UIntIntegerImpl$.minusContext(UIntTypeClass.scala:143)
[error] 	at dsptools.numbers.UIntImpl$UIntIntegerImpl$.minusContext(UIntTypeClass.scala:143)
[error] 	at dsptools.numbers.ContextualRingOps.context_$minus(AllOps.scala:128)
[error] 	at dsptools.numbers.BadUIntSubtractWithGrow2.$anonfun$r$2(NumbersSpec.scala:409)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:62)
[error] 	at dsptools.DspContext$.withOverflowType(DspContext.scala:69)
[error] 	at dsptools.numbers.BadUIntSubtractWithGrow2.<init>(NumbersSpec.scala:409)
[error] 	at dsptools.numbers.NumbersSpec.$anonfun$new$11(NumbersSpec.scala:50)
[error] 	... (Stack trace trimmed to user code only, rerun with --full-stacktrace if you wish to see the full stack trace)

null was not equal to dsptools.DspException: OverflowType Grow is not supported for UInt subtraction
ScalaTestFailureLocation: dsptools.numbers.NumbersSpec at (NumbersSpec.scala:56)
Expected :dsptools.DspException: OverflowType Grow is not supported for UInt subtraction
Actual   :null
<Click to see difference>

org.scalatest.exceptions.TestFailedException: null was not equal to dsptools.DspException: OverflowType Grow is not supported for UInt subtraction
	at org.scalatest.MatchersHelper$.indicateFailure(MatchersHelper.scala:343)
	at org.scalatest.Matchers$ShouldMethodHelper$.shouldMatcher(Matchers.scala:6723)
	at org.scalatest.Matchers$AnyShouldWrapper.should(Matchers.scala:6759)
	at dsptools.numbers.NumbersSpec.$anonfun$new$9(NumbersSpec.scala:56)
	at org.scalatest.OutcomeOf.outcomeOf(OutcomeOf.scala:85)
	at org.scalatest.OutcomeOf.outcomeOf$(OutcomeOf.scala:83)
	at org.scalatest.OutcomeOf$.outcomeOf(OutcomeOf.scala:104)
	at org.scalatest.Transformer.apply(Transformer.scala:22)
	at org.scalatest.Transformer.apply(Transformer.scala:20)
	at org.scalatest.FreeSpecLike$$anon$1.apply(FreeSpecLike.scala:405)
	at org.scalatest.TestSuite.withFixture(TestSuite.scala:196)
	at org.scalatest.TestSuite.withFixture$(TestSuite.scala:195)
	at org.scalatest.FreeSpec.withFixture(FreeSpec.scala:1739)
	at org.scalatest.FreeSpecLike.invokeWithFixture$1(FreeSpecLike.scala:403)
	at org.scalatest.FreeSpecLike.$anonfun$runTest$1(FreeSpecLike.scala:415)
	at org.scalatest.SuperEngine.runTestImpl(Engine.scala:286)
	at org.scalatest.FreeSpecLike.runTest(FreeSpecLike.scala:415)
	at org.scalatest.FreeSpecLike.runTest$(FreeSpecLike.scala:397)
	at org.scalatest.FreeSpec.runTest(FreeSpec.scala:1739)
	at org.scalatest.FreeSpecLike.$anonfun$runTests$1(FreeSpecLike.scala:474)
	at org.scalatest.SuperEngine.$anonfun$runTestsInBranch$1(Engine.scala:393)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at org.scalatest.SuperEngine.traverseSubNodes$1(Engine.scala:381)
	at org.scalatest.SuperEngine.runTestsInBranch(Engine.scala:370)
	at org.scalatest.SuperEngine.$anonfun$runTestsInBranch$1(Engine.scala:407)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at org.scalatest.SuperEngine.traverseSubNodes$1(Engine.scala:381)
	at org.scalatest.SuperEngine.runTestsInBranch(Engine.scala:370)
	at org.scalatest.SuperEngine.$anonfun$runTestsInBranch$1(Engine.scala:407)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at org.scalatest.SuperEngine.traverseSubNodes$1(Engine.scala:381)
	at org.scalatest.SuperEngine.runTestsInBranch(Engine.scala:370)
	at org.scalatest.SuperEngine.$anonfun$runTestsInBranch$1(Engine.scala:407)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at org.scalatest.SuperEngine.traverseSubNodes$1(Engine.scala:381)
	at org.scalatest.SuperEngine.runTestsInBranch(Engine.scala:376)
	at org.scalatest.SuperEngine.runTestsImpl(Engine.scala:458)
	at org.scalatest.FreeSpecLike.runTests(FreeSpecLike.scala:474)
	at org.scalatest.FreeSpecLike.runTests$(FreeSpecLike.scala:473)
	at org.scalatest.FreeSpec.runTests(FreeSpec.scala:1739)
	at org.scalatest.Suite.run(Suite.scala:1124)
	at org.scalatest.Suite.run$(Suite.scala:1106)
	at org.scalatest.FreeSpec.org$scalatest$FreeSpecLike$$super$run(FreeSpec.scala:1739)
	at org.scalatest.FreeSpecLike.$anonfun$run$1(FreeSpecLike.scala:518)
	at org.scalatest.SuperEngine.runImpl(Engine.scala:518)
	at org.scalatest.FreeSpecLike.run(FreeSpecLike.scala:518)
	at org.scalatest.FreeSpecLike.run$(FreeSpecLike.scala:517)
	at org.scalatest.FreeSpec.run(FreeSpec.scala:1739)
	at org.scalatest.tools.SuiteRunner.run(SuiteRunner.scala:45)
	at org.scalatest.tools.Runner$.$anonfun$doRunRunRunDaDoRunRun$13(Runner.scala:1349)
	at org.scalatest.tools.Runner$.$anonfun$doRunRunRunDaDoRunRun$13$adapted(Runner.scala:1343)
	at scala.collection.immutable.List.foreach(List.scala:392)
	at org.scalatest.tools.Runner$.doRunRunRunDaDoRunRun(Runner.scala:1343)
[... cut ...]
	at org.scalatest.tools.Runner$.$anonfun$runOptionallyWithPassFailReporter$24$adapted(Runner.scala:1011)
	at org.scalatest.tools.Runner$.withClassLoaderAndDispatchReporter(Runner.scala:1509)
	at org.scalatest.tools.Runner$.runOptionallyWithPassFailReporter(Runner.scala:1011)
	at org.scalatest.tools.Runner$.run(Runner.scala:850)
	at org.scalatest.tools.Runner.run(Runner.scala)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.runScalaTest2(ScalaTestRunner.java:133)
	at org.jetbrains.plugins.scala.testingSupport.scalaTest.ScalaTestRunner.main(ScalaTestRunner.java:27)
```

seldridge, 2020-03-21:
I think there's two options here:

1. `ChiselStage` should include the `ChiselException` as the cause of the `StageError`
1. `ChiselStage` should set the `StageError` cause to the cause of the `ChiselException`

(1) seems like the right way to go, but will require a small change to the test here (there are then two causes to unwrap). (2) seems like it's hiding information but should cause the test to work as written.

jackkoenig, 2022-03-22:
Fixed by https://github.com/chipsalliance/chisel3/pull/1382

closed 2022-03-22 by jackkoenig