chipsalliance/chisel #1117: ExtModule needs utils like BlackBox has

chick, 2019-06-26:
## Type of Issue: Code completeness, Testers support
BlackBoxUtils provide an annotation system for retrieving blackbox verilog code
from resource files. Specifically the ability to declare in-line verilog and verilog as resource.
ExtMode does not provide a similar mechanism.

## Impact
Lack of this support leads to ad hoc system for placement of verilog code, and makes
writing libraries that rely on black boxed verilog more complicated. It also complicates
testers implementation of black box testing.

## Development required

- Add ExtModule utils similar to BlackBoxUtils
- Make sure Treadle and testers take advantage of these tools
- Testers2 and Treadle and possibly testers need to be updated to make it possible to write tests with black boxes that work with multiple backends

## Use case
If we do not provide consistent implementation tools and decent documentation on how to use them. The ecosystem ends up with uncontrolled mechanism that are likely to break as we issue new code.

ducky64, 2019-06-26:
Thoughts about maybe writing this to be less mutable (`addResource` / `setResource` / `setInline`) and more declarative (maybe we could have `class InlineExtModule` / `class ResourceExtModule(resourcePath)` - or making ExtModule directly take a resource path as an optional constructor parameter) or Chisel-like (not sure what this would look like, since the ExtModule body only has IO defs)? The semantics probably are fine as-is, but the syntax I think could be more uniform.

For inline Verilog, would it make sense to have that as a feature that could be used in a regular module (like inline assembly in C)? It would seem that if you're writing large chunks of Verilog, that should go into a `.v` file, and if you're writing small chunks, you maybe just want to (or even need to) inline it into the instantiating module.

azidar, 2020-10-27:
Done.

closed 2020-10-27 by azidar