chipsalliance/chisel #1119: Assign Clock equal to 1

Nguyendien, 2019-07-02:
In verilog, I can assign Clock is equal to another clock or 1. But I can't assign clock is 1.U in chisel? How can I fix it?
Verilog:
 assign  clk_IR      = ((state_reg  == SHIFT_IR)||(state_reg == CAPTURE_IR)) ? clock : 1'b1;
Chisel3:
  when(state_reg === SHIFT_IR || state_reg ===  CAPTURE_IR){
    io.clk_IR := clock
  }.otherwise{
    io.clk_IR := 1.U // Error cannot connect chisel3.core.Clock@2c and chisel3.core.UInt@54
  }
Thanks!

sequencer, 2019-07-02:
This is a ClockMux, which is not supported in firrtl.
And this use case is rare in verilog too, only carefully consider what circuit should use and add correct constraints can build the circuit you needed.
If you are design a Xilinx FPGA circuit, you might need BUFGMUX from http://www.xilinx.com/support/documentation/user_guides/ug382.pdf 
If you are design a ASIC, stdcell will have this specific mux for clock, and you need to map it in synthesis tools.

edwardcwang, 2019-07-02:
In addition to @sequencer's comment about clock muxes, if you really wanted to just assign a clock high, you could do `true.B.asClock`. The statement clock mux advisory still applies.

Nguyendien, 2019-07-03:
@sequencer @edwardcwang Can you tell me how can I generate code verilog
`assign clk_IR = ((state_reg == SHIFT_IR)||(state_reg == CAPTURE_IR)) ? clock : 1'b1;`
to chisel3 which I can gen firrtl and verilog?

closed 2022-03-22 by jackkoenig