chipsalliance/chisel #1121: UNRESOLVED DEPENDENCIES

mlulaj, 2019-07-09:
**Type of issue**: I am trying to compile this github project : https://github.com/ekiwi/rfuzz
And I am facing an error. Im am behind a corporate proxy. When I try to make run on the pc of the corporate  get the foll
``
owing error:

I tried to compile the project on my personal pc, connected on another wifi, not the one of the corporate and it is running correctly. But on the  corporate pc it gives the error

**What is the expected behavior?**
It should  compile without problems

I tried to use another version of sbt and scala , but is the same . 
I tried sbt update
ERROR LOG :
` make run
cd chisel3 ;\
  sbt -ivy /home/mustafa/Desktop/rfuzz2ndtry/.ivy2 publishLocal && touch /home/mustafa/Desktop/rfuzz2ndtry/chisel.stamp
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/mustafa/Desktop/rfuzz2ndtry/chisel3/project
[info] Loading settings from build.sbt ...
[info] Set current project to chisel3 (in build file:/home/mustafa/Desktop/rfuzz2ndtry/chisel3/)
[info] Packaging /home/mustafa/Desktop/rfuzz2ndtry/chisel3/target/scala-2.11/chisel3_2.11-1.2-SNAPSHOT-sources.jar ...
[info] Done packaging.
[info] Updating coreMacros...
[warn] 	module not found: edu.berkeley.cs#firrtl_2.11;1.2-SNAPSHOT
[warn] 	Detected merged artifact: [FAILED     ] org.scala-lang#scala-library;2.11.12!scala-library.jar:  (0ms).
[warn] 	Detected merged artifact: [FAILED     ] org.scala-lang#scala-reflect;2.11.12!scala-reflect.jar:  (0ms).
[warn] 	Detected merged artifact: [FAILED     ] org.scalamacros#paradise_2.11.12;2.1.0!paradise_2.11.12.jar:  (0ms).
[warn] 	Detected merged artifact: [FAILED     ] org.scala-lang#scala-compiler;2.11.12!scala-compiler.jar:  (0ms).
[warn] 	Detected merged artifact: [FAILED     ] org.scala-lang.modules#scala-xml_2.11;1.0.5!scala-xml_2.11.jar(bundle):  (0ms).
[warn] 	Detected merged artifact: [FAILED     ] org.scala-lang.modules#scala-parser-combinators_2.11;1.0.4!scala-parser-combinators_2.11.jar(bundle):  (0ms).
[warn] 	Detected merged artifact: [FAILED     ] jline#jline;2.14.3!jline.jar:  (0ms).
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: edu.berkeley.cs#firrtl_2.11;1.2-SNAPSHOT: not found
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::              FAILED DOWNLOADS            ::
[warn] 	:: ^ see resolution messages for details  ^ ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: org.scala-lang#scala-library;2.11.12!scala-library.jar
[warn] 	:: org.scala-lang#scala-reflect;2.11.12!scala-reflect.jar
[warn] 	:: org.scalamacros#paradise_2.11.12;2.1.0!paradise_2.11.12.jar
[warn] 	:: org.scala-lang#scala-compiler;2.11.12!scala-compiler.jar
[warn] 	:: org.scala-lang.modules#scala-xml_2.11;1.0.5!scala-xml_2.11.jar(bundle)
[warn] 	:: org.scala-lang.modules#scala-parser-combinators_2.11;1.0.4!scala-parser-combinators_2.11.jar(bundle)
[warn] 	:: jline#jline;2.14.3!jline.jar
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 
[warn] 	Note: Unresolved dependencies path:
[warn] 		edu.berkeley.cs:firrtl_2.11:1.2-SNAPSHOT
[warn] 		  +- edu.berkeley.cs:coremacros_2.11:1.2-SNAPSHOT
[info] Wrote /home/mustafa/Desktop/rfuzz2ndtry/chisel3/target/scala-2.11/chisel3_2.11-1.2-SNAPSHOT.pom
[error] sbt.librarymanagement.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.2-SNAPSHOT: not found
[error] download failed: org.scala-lang#scala-library;2.11.12!scala-library.jar
[error] download failed: org.scala-lang#scala-reflect;2.11.12!scala-reflect.jar
[error] download failed: org.scalamacros#paradise_2.11.12;2.1.0!paradise_2.11.12.jar
[error] download failed: org.scala-lang#scala-compiler;2.11.12!scala-compiler.jar
[error] download failed: org.scala-lang.modules#scala-xml_2.11;1.0.5!scala-xml_2.11.jar(bundle)
[error] download failed: org.scala-lang.modules#scala-parser-combinators_2.11;1.0.4!scala-parser-combinators_2.11.jar(bundle)
[error] download failed: jline#jline;2.14.3!jline.jar
[error] 	at sbt.internal.librarymanagement.IvyActions$.resolveAndRetrieve(IvyActions.scala:331)
[error] 	at sbt.internal.librarymanagement.IvyActions$.$anonfun$updateEither$1(IvyActions.scala:205)
[error] 	at sbt.internal.librarymanagement.IvySbt$Module.$anonfun$withModule$1(Ivy.scala:243)
[error] 	at sbt.internal.librarymanagement.IvySbt.$anonfun$withIvy$1(Ivy.scala:204)
[error] 	at sbt.internal.librarymanagement.IvySbt.sbt$internal$librarymanagement$IvySbt$$action$1(Ivy.scala:70)
[error] 	at sbt.internal.librarymanagement.IvySbt$$anon$3.call(Ivy.scala:77)
[error] 	at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:95)
[error] 	at xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:80)
[error] 	at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:99)
[error] 	at xsbt.boot.Using$.withResource(Using.scala:10)
[error] 	at xsbt.boot.Using$.apply(Using.scala:9)
[error] 	at xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:60)
[error] 	at xsbt.boot.Locks$GlobalLock.withLock(Locks.scala:50)
[error] 	at xsbt.boot.Locks$.apply0(Locks.scala:31)
[error] 	at xsbt.boot.Locks$.apply(Locks.scala:28)
[error] 	at sbt.internal.librarymanagement.IvySbt.withDefaultLogger(Ivy.scala:77)
[error] 	at sbt.internal.librarymanagement.IvySbt.withIvy(Ivy.scala:199)
[error] 	at sbt.internal.librarymanagement.IvySbt.withIvy(Ivy.scala:196)
[error] 	at sbt.internal.librarymanagement.IvySbt$Module.withModule(Ivy.scala:242)
[error] 	at sbt.internal.librarymanagement.IvyActions$.updateEither(IvyActions.scala:190)
[error] 	at sbt.librarymanagement.ivy.IvyDependencyResolution.update(IvyDependencyResolution.scala:20)
[error] 	at sbt.librarymanagement.DependencyResolution.update(DependencyResolution.scala:56)
[error] 	at sbt.internal.LibraryManagement$.resolve$1(LibraryManagement.scala:46)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$12(LibraryManagement.scala:99)
[error] 	at sbt.util.Tracked$.$anonfun$lastOutput$1(Tracked.scala:68)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$19(LibraryManagement.scala:112)
[error] 	at scala.util.control.Exception$Catch.apply(Exception.scala:224)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11(LibraryManagement.scala:112)
[error] 	at sbt.internal.LibraryManagement$.$anonfun$cachedUpdate$11$adapted(LibraryManagement.scala:95)
[error] 	at sbt.util.Tracked$.$anonfun$inputChanged$1(Tracked.scala:149)
[error] 	at sbt.internal.LibraryManagement$.cachedUpdate(LibraryManagement.scala:126)
[error] 	at sbt.Classpaths$.$anonfun$updateTask$5(Defaults.scala:2383)
[error] 	at scala.Function1.$anonfun$compose$1(Function1.scala:44)
[error] 	at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:39)
[error] 	at sbt.std.Transform$$anon$4.work(System.scala:66)
[error] 	at sbt.Execute.$anonfun$submit$2(Execute.scala:262)
[error] 	at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:16)
[error] 	at sbt.Execute.work(Execute.scala:271)
[error] 	at sbt.Execute.$anonfun$submit$1(Execute.scala:262)
[error] 	at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:174)
[error] 	at sbt.CompletionService$$anon$2.call(CompletionService.scala:36)
[error] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[error] 	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
[error] 	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
[error] 	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
[error] 	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
[error] 	at java.base/java.lang.Thread.run(Thread.java:834)
[error] (coreMacros / update) sbt.librarymanagement.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.2-SNAPSHOT: not found
[error] download failed: org.scala-lang#scala-library;2.11.12!scala-library.jar
[error] download failed: org.scala-lang#scala-reflect;2.11.12!scala-reflect.jar
[error] download failed: org.scalamacros#paradise_2.11.12;2.1.0!paradise_2.11.12.jar
[error] download failed: org.scala-lang#scala-compiler;2.11.12!scala-compiler.jar
[error] download failed: org.scala-lang.modules#scala-xml_2.11;1.0.5!scala-xml_2.11.jar(bundle)
[error] download failed: org.scala-lang.modules#scala-parser-combinators_2.11;1.0.4!scala-parser-combinators_2.11.jar(bundle)
[error] download failed: jline#jline;2.14.3!jline.jar
[error] Total time: 2 s, completed Jul 9, 2019, 4:18:58 PM
Makefile:65: recipe for target '/home/mustafa/Desktop/rfuzz2ndtry/chisel.stamp' failed
`

wachag, 2019-07-09:
Have you tried contacting the authors of the project?

mlulaj <notifications@github.com> ezt írta (időpont: 2019. júl. 9., K
16:44):

ucbjrl, 2019-07-09:
It looks like your proxy isn't downloading published Sonatype/Maven artifacts from the internet. Verify the `build.sbt` for the project contains the following:
```scala
resolvers ++= Seq(
  Resolver.sonatypeRepo("snapshots"),
  Resolver.sonatypeRepo("releases")
)
```
If that doesn't fix the problem, it's probably a policy or a proxy configuration issue.

You might try the following, which should work _out of the box_:
```bash
$ git clone https://github.com/freechipsproject/chisel-template.git
$ cd chisel-template
$ sbt clean test
```
You should something like the following in the output:
```
[info] Updating ...
[warn] Choosing sonatype-snapshots for edu.berkeley.cs#chisel3_2.12;3.2-SNAPSHOT
[warn] Choosing sonatype-snapshots for edu.berkeley.cs#firrtl_2.12;1.2-SNAPSHOT
[warn] Choosing sonatype-snapshots for edu.berkeley.cs#chisel-iotesters_2.12;1.3
-SNAPSHOT
[warn] Choosing sonatype-snapshots for edu.berkeley.cs#firrtl-interpreter_2.12;1
.2-SNAPSHOT
[warn] Choosing sonatype-snapshots for edu.berkeley.cs#treadle_2.12;1.1-SNAPSHOT
[warn] Choosing sonatype-snapshots for edu.berkeley.cs#chisel-testers2_2.12;0.1-
SNAPSHOT
[info] downloading https://oss.sonatype.org/content/repositories/snapshots/edu/b
erkeley/cs/firrtl_2.12/1.2-SNAPSHOT/firrtl_2.12-1.2-SNAPSHOT.jar ...
[info] downloading https://oss.sonatype.org/content/repositories/snapshots/edu/b
erkeley/cs/chisel-iotesters_2.12/1.3-SNAPSHOT/chisel-iotesters_2.12-1.3-SNAPSHOT
.jar ...
[info] downloading https://oss.sonatype.org/content/repositories/snapshots/edu/b
erkeley/cs/chisel-testers2_2.12/0.1-SNAPSHOT/chisel-testers2_2.12-0.1-SNAPSHOT.j
ar ...
[info] downloading https://oss.sonatype.org/content/repositories/snapshots/edu/b
erkeley/cs/firrtl-interpreter_2.12/1.2-SNAPSHOT/firrtl-interpreter_2.12-1.2-SNAP
SHOT.jar ...
[info] downloading https://oss.sonatype.org/content/repositories/snapshots/edu/b
erkeley/cs/chisel3_2.12/3.2-SNAPSHOT/chisel3_2.12-3.2-SNAPSHOT.jar ...
[info] downloading https://oss.sonatype.org/content/repositories/snapshots/edu/b
erkeley/cs/treadle_2.12/1.1-SNAPSHOT/treadle_2.12-1.1-SNAPSHOT.jar ...
[info]  [SUCCESSFUL ] edu.berkeley.cs#chisel-iotesters_2.12;1.3-SNAPSHOT!chisel-
iotesters_2.12.jar (2532ms)
[info]  [SUCCESSFUL ] edu.berkeley.cs#firrtl-interpreter_2.12;1.2-SNAPSHOT!firrt
l-interpreter_2.12.jar (2553ms)
[info]  [SUCCESSFUL ] edu.berkeley.cs#chisel3_2.12;3.2-SNAPSHOT!chisel3_2.12.jar
 (2609ms)
[info]  [SUCCESSFUL ] edu.berkeley.cs#treadle_2.12;1.1-SNAPSHOT!treadle_2.12.jar
 (2646ms)
[info]  [SUCCESSFUL ] edu.berkeley.cs#chisel-testers2_2.12;0.1-SNAPSHOT!chisel-t
esters2_2.12.jar (2704ms)
[info]  [SUCCESSFUL ] edu.berkeley.cs#firrtl_2.12;1.2-SNAPSHOT!firrtl_2.12.jar (
2732ms)
[info] Done updating.
[info] Compiling 2 Scala sources to /nscratch/ucbjrl/chisel-template/target/scal
a-2.12/classes ...

```

mlulaj, 2019-07-10:
@wachag  Yes I already contacted them but no reply yet.  Im trying to fix this error since 1 week, and still no result.

mlulaj, 2019-07-10:
@ucbjrl I checked on build.sbt file and these lines are there.

The second one, gives me error too: 

 sbt clean test
[info] Loading settings from plugins.sbt ...
[info] Loading project definition from /home/mustafa/Desktop/chisel-template/project
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	::          UNRESOLVED DEPENDENCIES         ::
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 	:: org.scala-lang#scala-library;2.12.4: Resolution failed several times for dependency: org.scala-lang#scala-library;2.12.4 {scala-tool=[default, optional(default)], provided=[default(compile)]}:: 
[warn] 	typesafe-ivy-releases: unable to get resource for org.scala-lang#scala-library;2.12.4: res=https://repo.typesafe.com/typesafe/ivy-releases/org.scala-lang/scala-library/2.12.4/ivys/ivy.xml: javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	sbt-plugin-releases: unable to get resource for org.scala-lang#scala-library;2.12.4: res=https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/org.scala-lang/scala-library/2.12.4/ivys/ivy.xml: javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	:: org.scala-sbt#sbt;1.1.1: Resolution failed several times for dependency: org.scala-sbt#sbt;1.1.1 {provided=[default(compile)]}:: 
[warn] 	typesafe-ivy-releases: unable to get resource for org.scala-sbt#sbt;1.1.1: res=https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/sbt/1.1.1/ivys/ivy.xml: javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	sbt-plugin-releases: unable to get resource for org.scala-sbt#sbt;1.1.1: res=https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/org.scala-sbt/sbt/1.1.1/ivys/ivy.xml: javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	:: org.scala-lang#scala-compiler;2.12.4: Resolution failed several times for dependency: org.scala-lang#scala-compiler;2.12.4 {scala-tool=[default, optional(default)]}:: 
[warn] 	typesafe-ivy-releases: unable to get resource for org.scala-lang#scala-compiler;2.12.4: res=https://repo.typesafe.com/typesafe/ivy-releases/org.scala-lang/scala-compiler/2.12.4/ivys/ivy.xml: javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	sbt-plugin-releases: unable to get resource for org.scala-lang#scala-compiler;2.12.4: res=https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/org.scala-lang/scala-compiler/2.12.4/ivys/ivy.xml: javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[warn] 	::::::::::::::::::::::::::::::::::::::::::::::
[warn] 
[warn] 	Note: Unresolved dependencies path:
[warn] 		org.scala-lang:scala-library:2.12.4 ((sbt.Classpaths.jvmBaseSettings) Defaults.scala#L2055)
[warn] 		  +- default:chisel-template-build:0.1.0-SNAPSHOT (scalaVersion=2.12, sbtVersion=1.0)
[warn] 		org.scala-sbt:sbt:1.1.1
[warn] 		  +- default:chisel-template-build:0.1.0-SNAPSHOT (scalaVersion=2.12, sbtVersion=1.0)
[warn] 		org.scala-lang:scala-compiler:2.12.4
[warn] 		  +- default:chisel-template-build:0.1.0-SNAPSHOT (scalaVersion=2.12, sbtVersion=1.0)
[error] sbt.librarymanagement.ResolveException: unresolved dependency: org.scala-lang#scala-library;2.12.4: Resolution failed several times for dependency: org.scala-lang#scala-library;2.12.4 {scala-tool=[default, optional(default)], provided=[default(compile)]}:: 
[error] 	typesafe-ivy-releases: unable to get resource for org.scala-lang#scala-library;2.12.4: res=https://repo.typesafe.com/typesafe/ivy-releases/org.scala-lang/scala-library/2.12.4/ivys/ivy.xml: javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[error] 	sbt-plugin-releases: unable to get resource for org.scala-lang#scala-library;2.12.4: res=https://repo.scala-sbt.org/scalasbt/sbt-plugin-releases/org.scala-lang/scala-library/2.12.4/ivys/ivy.xml: javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[error] 
[error] unresolved dependency: org.scala-sbt#sbt;1.1.1: Resolution failed several times for dependency: org.scala-sbt#sbt;1.1.1 {provided=[default(compile)]}:: 
[error] 	typesafe-ivy-releases: unable to get resource for org.scala-sbt#sbt;1.1.1: res=https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/sbt/1.1.1/ivys/ivy.xml: javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
[... cut ...]
It is a wierd situation, I dont think its only the security of internet which I am connected to. 
I tired to connect it on my personal wifi, and its doing the same . Im afraid its something regarding the security of pc.
This pc is configured to work on the wifi of the company only.  When I connect using other wifi, I can not access the internet using WINDOWS, but only on virtual machine, by disabling the proxy configurations.

If you have any idea about this, you would really help me because Im stucked really bad on this. 
Thanks

ucbjrl, 2019-07-10:
You'll need assistance from your company networking support to resolve this. I don't think there's anything we can do from the outside.

ducky64, 2019-07-10:
This seems like a corporate proxy issue which is preventing sbt from downloading the necessary JARs.

You might also want to check that your JDK/JRE is up to date - it caused the certification errors for me, but that doesn't sound like the case for you.

I think we created an offline distribution of Chisel3 a long time ago for use in one of the bootcamp days or network-inaccessible environments. I don't know if that's something we're still maintaining, but that should bypass this problem completely.

mlulaj, 2019-07-11:
I managed to bypass this error. Yes, it was the proxy which didn't allow me to download anything , so I created a local repository , putting inside every file it wanted, copying from another machine , on which these files were already downloaded. 
Now its running correctly. 
Thank you for your support :)

closed 2019-07-11 by mlulaj