chipsalliance/chisel #1122: About Mem Port.

245950258, 2019-07-12:
I use SeqMem to generate a ram with a white mask. And I add --repl-seq-mem this command to Replace sequential memories with black boxes.
However, the clock, en and address port are divided into reading and writing as the following codes.
`ram ram ( // @[DescribedSRAM.scala 20:21]
    .R0_addr(data_arrays_0_R0_addr),
    .R0_en(data_arrays_0_R0_en),
    .R0_clk(data_arrays_0_R0_clk),
    .R0_data_0(data_arrays_0_R0_data_0),
    .R0_data_1(data_arrays_0_R0_data_1),
    .R0_data_2(data_arrays_0_R0_data_2),
    .R0_data_3(data_arrays_0_R0_data_3),
    .W0_addr(data_arrays_0_W0_addr),
    .W0_en(data_arrays_0_W0_en),
    .W0_clk(data_arrays_0_W0_clk),
    .W0_data_0(data_arrays_0_W0_data_0),
    .W0_data_1(data_arrays_0_W0_data_1),
    .W0_data_2(data_arrays_0_W0_data_2),
    .W0_data_3(data_arrays_0_W0_data_3),
    .W0_mask_0(data_arrays_0_W0_mask_0),
    .W0_mask_1(data_arrays_0_W0_mask_1),
    .W0_mask_2(data_arrays_0_W0_mask_2),
    .W0_mask_3(data_arrays_0_W0_mask_3)
  );`

What should I do, If I want to generate a ram like this.
`module ram(
  input RW0_clk,
  input [9:0] RW0_addr,
  input RW0_en,
  input RW0_wmode,
  input [3:0] RW0_wmask,
  input [31:0] RW0_wdata,
  output [31:0] RW0_rdata
);`

edwardcwang, 2019-07-12:
You can take a look at https://github.com/freechipsproject/chisel3/wiki/Memories#mem to see how to infer a single-port read-write memory.

245950258, 2019-07-12:
You are right.
I find that I can add the "-infer-rw" command to generate the single port ram without changing codes while executing chisel.
Thanks a lot.

closed 2019-07-12 by 245950258