chipsalliance/chisel #1125: Type after shift in IntelliJ

schoeberl, 2019-07-17:
Maybe this is just a bug in IntelliJ, but let's see if someone has an idea.

**Type of issue**: bug report

Following code
```Scala
  val a = 1.U << 1
  val b = 3.U
  println(a, b)

  // IntelliJ editor complains on 'a': Type mismatch, expected UInt, actual: Bits
  when(b >= a) {
    io.data := 1.U
  }
```

displays an error in the IntelliJ editor (a is marked in the when expression).  The editor thinks that the shift operation in the first line results in Bits and not a UInt.

However, the code compiles correctly with IntelliJ and also with sbt from the terminal.

What, in the Chisel type system, is confusing the IntelliJ editor here?

**Please tell us about your environment:**
-version 3.1.8
- IntelliJ latest on Mac (just updated)

kammoh, 2019-07-17:
*UInt* is a subtype of the class *BIts*, where `<<` is defined with a return type of Bits. So IntelliJ is somewhat right about that! As I understand it, what makes this actually work at compile time is the fact that `<<` is that defined as a macro, and with the proper macro expansion the correct `do_<<` implementation with the correct return type is selected (so maybe it's also fair to attribute this problem to the realtime macro expansions in IntelliJ). IIRC the same issue could arise with `>>` and `~`, for UInts,  FixedPoint, and Bool. 

This is probably just a small nuisance (or again IntelliJ could do better with the macro expansions), and can be resolved by adding a (redundant) `asUInt` at the outcome of these operators. 
But as a side-note, I actually do find the type hierarchy in Chisel (and its implementation) a little over-convoluted and deserving of some attention/revisit. At the same time, I can’t really think of an easy fix for this particular issue. 
Type classes are typically used in these situations and UInt (and SInt) already extends the type class `Num[T]`. Maybe we can argue that shift operators are in fact arithmetic operations and can go in `Num[T]` and return `T` (which resolves this issue). But the problem is that the bit extraction operator in Bits (definitely not arithmetic) is implemented using `>>`.

kammoh, 2019-07-18:
One possible workaround is to add an implicit conversion method in scope (possibly in your project's package object). Note that implicit conversion methods are generally frowned upon, but there shouldn't be much risk involved in using this in the Chisel context.

```scala
import scala.language.implicitConversions
implicit def bitsToUInt(b: Bits): UInt = b.asUInt
```

schoeberl, 2019-07-19:
I was expecting that this is an issue with the Chisel type system being a little bit too complicated for IntelliJ. I know this is not an Chisel error, but I think this is an issue for Chisel.

I don't think this workaround is a good idea. We don't want to be able to always have this conversion.

If we want to get Chisel mainstream accepted, we need to be able to use Scala IDEs such as ItelliJ or Eclipse. For example, I want to use Chisel next semester in teaching. How shall I explain those students that some errors in the editor are ok to ignore?

Cheers,
Martin

schoeberl, 2019-07-19:
BTW, the same issue is true with Eclipse :-(

And another example:
```
  val blkReg = Reg(UInt(1.W))

  cntReg := cntReg + 1.U
  when(cntReg === CNT_MAX) {
    cntReg := 0.U
    blkReg := ~blkReg
  }
  io.led := blkReg
}
```

on blkReg := ~blkReg it complains with a type mismatch.

ducky64, 2019-07-19:
Implicit conversions: there's been some discussion on making the separation of Bits and UInt better, so it might not be safe to have Bits automatically cast to UInt.

I think a typeclass solution on Bits makes sense, and I remember hearing that having macros change the return type was discouraged and may go away in a future Scala version. One caveat here is that it needs to not bleed into the user facing API (this is what prevented a typeclass approach to static-typed literals - it would have required something like `MyBundle extends Bundle with StaticLiteral[MyBundle]`), but since this only affects the Bits hierarchy (which we don't expect users to extend), that shouldn't be a problem

schoeberl, 2023-01-12:
I want to bring this issue up. It bites me over and over again that a shift converts a UInt or SInt to Bits. I forgot why this is the case. But it is super annoying. Can we simply define the shift operators on UInt and SInt?