chipsalliance/chisel #1130: Reference io is not fully initialized

Nguyendien, 2019-07-25:
`    when(io.updateDR){
      io.wdata := user_reg
    }`
when I convert chisel to verilog. Error: **Reference io is not fully initialized**, because no .otherwise. But I only need to update wdata when updateDR = 1. If updateDR = 0, wdata keep value?
Thanks!

edwardcwang, 2019-07-25:
`io.wdata` is a wire, so it doesn't have the concept of "keeping the same value". When we use `when()` to assign to wires, we are creating muxes.

Nguyendien, 2019-07-25:
how can I generate RTL verilog `if(updateDR) begin wdata = user_reg; end` to chisel 3 if I dont use when statement?

edwardcwang, 2019-07-25:
Do you want this in an `always @ (posedge clock)` block?

Nguyendien, 2019-07-25:
No, it is not depend on clock

edwardcwang, 2019-07-25:
What is the hardware you are trying to build? A register or a combinational mux?

Nguyendien, 2019-07-25:
it is a combinational mux

edwardcwang, 2019-07-25:
Let's say you want to build a mux in Chisel that muxes with `sel` between `s0` and `s1` and assigns the output to `out`:
```scala
when (sel) {
  out := s1
} .otherwise {
  out := s0
}
```

Nguyendien, 2019-07-26:
No, I mean that:
`when (sel) {
out := s1
} .otherwise {
out := out
}`// but it is error "combinational loop detected"
This is similar to verilog
`if (sel)
out = s1;`

edwardcwang, 2019-07-26:
Do you want to create a latch in that case? (a circuit whose state is updated when `sel` is high)?

Nguyendien, 2019-07-26:
yes, @edwardcwang

edwardcwang, 2019-07-26:
I see. Currently, Chisel does not have a native construct for latches. If it is needed for your design, we can create a Blackbox module for a latch and instantiate/use it in Chisel: https://github.com/freechipsproject/chisel3/wiki/Blackboxes#blackboxes-with-in-line-verilog

Nguyendien, 2019-07-26:
thanks, but I have another question. In verilog, we have:
always @(negedge clk_DR or posedge captureDR) if (captureDR) ID_Reg <= 0x3;
. So how about chisel code look like ? How can detect change signal similar to verilog?

edwardcwang, 2019-07-26:
Glad to hear you were able to resolve it! I noticed there is another issue (#1131), so let's redirect the discussion there instead.

closed 2019-07-26 by edwardcwang