chipsalliance/chisel #1131: Posedge and negedge signal

Nguyendien, 2019-07-25:
In verilog, I can use `@(negedge clock or posedge signalA)`. How can I do similar in chisel?

chick, 2019-07-25:
How about this as an example.
```scala
class NegReg extends Module {
  val io = IO(new Bundle {
    val negClockOut = Output(Bool())
    val negRegOut   = Output(UInt(16.W))
  })

  val negClock = (~ clock.asUInt).asBool.asClock

  val negReg = withClock(negClock) {
    RegInit(0.U(16.W))
  }
  negReg := negReg + 1.U

  io.negClockOut := negClock.asUInt.asBool
  io.negRegOut   := negReg
}

```

chick, 2019-07-25:
And thanks for your question but, if possible, questions like this are best asked and answered on stackoverflow or gitter.im/freechipsproject/chisel3. This makes the answers more accessible to the community at large. Issues are better for recording bugs or enhancement requests.

kammoh, 2019-07-25:
Nice example @chick. Would it be possible to include it in the Wiki for future references? And BTW what is the current overall status of #695?

Nguyendien, 2019-07-26:
In verilog, we have:
 `always @(negedge clk_DR or posedge captureDR)
if (captureDR) ID_Reg <= 0x3;
`
. So how about chisel code look like ?

sequencer, 2019-07-26:
firrtl doesn't have the clock edge information in the IR, thus `negedge` cannot be emitted by firrtl.
@chick gave a work around for this issue, but this also need to be carefully treated in synthesis EDA tools to generate clock tree.
these preliminary PR https://github.com/freechipsproject/firrtl/pull/1125 and https://github.com/freechipsproject/chisel3/pull/1127 give the clockedge information for firrtl, but also introduce a great change for AST, which need to be carefully treated.

Nguyendien, 2019-08-01:
I understand, but in verilog we have `always @(negedge clk_DR or posedge captureDR)`. This means we check the change of ###clk_DR or ###captureDR at one. In Chisel, I try `withClock(clk_DR or captureDR)`, it is error. How can I do this?