chipsalliance/chisel #1134: Queue.irrevocable

245950258, 2019-07-27:
I write the following codes in Chisel2. And it generates verilog fine.
`val io  = new Bundle  {
        val in  = Flipped(Decoupled(UInt(w.W)))
    }
val x = Queue.irrevocable(io.in, 4)
`

But when I try to convert this to chisel3, it reports "must be a Chisel type, not hardware".

I find that the problem is due to [https://github.com/freechipsproject/chisel3/blob/master/src/main/scala/chisel3/util/Decoupled.scala](url).

`private val genType = (DataMirror.internal.isSynthesizable(gen), chisel3.internal.Builder.currentModule) match {
    case (true, Some(module: MultiIOModule))
        if !module.compileOptions.declaredTypeMustBeUnbound => chiselTypeOf(gen)
    case _ => gen
  }`
If I change "!module.compileOptions.declaredTypeMustBeUnbound => chiselTypeOf(gen)" to “module.compileOptions.declaredTypeMustBeUnbound => chiselTypeOf(gen)”, I can get verilog without error.

edwardcwang, 2019-07-27:
Can you post your full Chisel3 module or a more complete representative Chisel3 example so that we can better help you?

245950258, 2019-07-27:
This is my code.
`import chisel3._
import chisel3.util._

class Queue5(w:Int , n:Int) extends Module{
    val io  = IO(new Bundle  {
        val in  = Flipped(Decoupled(UInt(w.W)))
        val out = Decoupled(UInt(w.W))
    })
    val x = Queue.irrevocable(io.in, n)
    x.ready := io.out.ready
    io.out.valid := x.valid
    io.out.bits := x.bits
} `

It is compiled successfully, But when elaborating design， reporting following error "'chisel3.core.UInt@79' must be a Chisel type, not hardware".

edwardcwang, 2019-07-28:
@245950258 Thank you for the catch - there does indeed appear to be a bug in Chisel. I've gotten a fix going in #1136 to fix it.

245950258, 2019-07-28:
It is my pleasure.

edwardcwang, 2019-07-28:
(leaving issue open for tracking)

commit e39525e9 (2019-07-28): Make Queue.irrevocable work properly in chisel3

Close #1134
commit 7d338606 (2019-07-28): Make Queue.irrevocable work properly in chisel3

Close #1134
commit 375e2297 (2019-07-28): Make Queue.irrevocable work properly in chisel3

Close #1134
commit 3a6ba0d4 (2020-02-10): Make Queue.irrevocable work properly in chisel3

Close #1134
closed 2020-02-10 by edwardcwang