chipsalliance/chisel #1135: Circular shift operator

kammoh, 2019-07-27:
Circular shift or bit rotation (https://en.wikipedia.org/wiki/Circular_shift) is common bit-wise operation in many applications including cryptography. VHDL offers operators `ror` and `rol` as well as more polished versions as functions `rotate_left` and `rotate_right` for signed and unsigned types. 
While the user can implement it using bit slicing and concatenation, it could become a little tricky and prone to mistakes. It would be nice to have operators/methods for this in Chisel.

A possible implementation for static rotation is:
```scala
  implicit class RotatableUInt(u: UInt) {
    def <<<(x: Int): UInt =
      if (x == 0)
        u
      else if (x > 0) {
        val w = u.getWidth
        u(w - x - 1, 0) ## u(w - 1, w - x)
      }
      else
        this >>> (-x)

    def >>>(x: Int): UInt =
      if (x > 0) {
        val w = u.getWidth
        u(x - 1, 0) ## u(w - 1, x)
      }
      else
        this <<< (-x)
```

Gotchas:
- Verilog and C don't seem to have operators for this operation. Verilog is using `>>>` operator as sign extension shift.

**Type of issue**: feature request

**Impact**: API addition (no impact on existing code)

**Development Phase**:  proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

aswaterman, 2019-07-27:
I'd avoid the names `<<<` and `>>>` like the plague.  Not only are they used for arithmetic shifts in Verilog, they're used for _logical_ shifts in Scala!  `rotateLeft` and `rotateRight` aren't too long to type, and they'll leave no one guessing.

The static rotates can be defined without relying on getWidth, so they can work without knowing the width at elaboration time:

    def rotateLeft(n: Int): UInt =
      if (n == 0) this
      else if (n < 0) this.rotateRight(-n)
      else this.tail(n) ## this.head(n)

    def rotateRight(n: Int): UInt =
      if (n <= 0) this.rotateLeft(-n)
      else this(n-1, 0)  ## (this >> n)

These will issue an error when the magnitude of n exceeds the width, which is an artificial limitation, but is probably OK in practice.

Supporting dynamic rotates is trickier because of argument reduction on the shift amount, given that we permit UInts whose width is not a power of 2.  What does it mean to right-shift a 13-bit-wide UInt by 9711?  The correct answer is, of course, that it should be equivalent to right-shifting it by (9711 % 13), or 0.  But computing that requires an expensive arithmetic circuit.  If you first truncate the rotate amount to 4 bits, you get a different result: (9711 % 16) % 13 is 2.

I would just recommend against defining dynamic rotates, or as a compromise, only define them for known-power-of-2-sized inputs, to save users from shooting themselves in the foot with either poor performance or surprising semantics.

kammoh, 2019-07-28:
Oh thanks @aswaterman! Now thinking about it, I'm absolutely with you on ditching those dubious symbols!  `rotateLeft` and `rotateRight` sound like reasonable choices.
It seems I had totally forgot about `tail` and `head` and their corresponding FIRRTL ops! This is obviously the right way to go. I think special case behavior might lead to confusion from the user's point of view. But maybe special-case failure would be an acceptable option? Fail in `rotate` (and not tail/head) when the width is known, otherwise fail in FIRRTL?
I also agree that only static rotates would make sense in order to keep the behavior consistent. VHDL only supports static versions as well.
Do you agree that these methods are best limited to `UInt`?
And would you mind if I churn out a PR based on your code and naming suggestions?

aswaterman, 2019-07-28:
I think we need to hear from some other chisel devs about expanding the
API, but it would be nice if you prepared the PR (and tests).

I’m fine with not having the special case, and erroring in firrtl if n is
too big. Other operators behave the same way.

Limiting this to UInt makes sense to me.

kammoh, 2019-08-05:
I've been using this operation in a couple of designs and so far I'm arriving at the conclusion that known-width modulo can probably be more sensible behavior in practice. The paradox is of course how can the designer reason about the behavior of these operations unless the width of the object is known to him.
I also added a low-cost modular dynamic barrel-shift based on the static rotates and updated the tests. I'd appreciate to hear your opinion.

aswaterman, 2019-08-05:
Yes, that implementation of dynamicShift is smarter than what I had
suggested.

Maybe dynamicShift should explicitly check that the width of the shift
amount is known, so it can print a more helpful error message, rather than
simply failing in asBools. It should also require the width of the shift
amount not exceed 31, else it will overflow the bounds of the Scala Int.

PR #1140 (2021-10-05): Circular-shift (rotate) operations for UInt
Adds *static* circular shift (rotation) operations `rotateLeft` and `rotateRight` for `UInt`.

**Related issue**: 
Closes #1135

**Type of change**: other enhancement

**Impact**: API addition (no impact on existing code)

**Development Phase**: implementation

**Release Notes**