chipsalliance/chisel #1137: Error generate code to RTL with blackbox

Nguyendien, 2019-07-30:
when I want to convert code chisel to verilog with black box, I have error : 

> [error] /data/workspace/chisel/chisel3-3.1.8/src/main/scala/tap/dti_bypass_register.scala:45:18: overloaded method value execute with alternatives:
[error]   (args: Array[String],dut: () => chisel3.experimental.RawModule)chisel3.ChiselExecutionResult <and>
[error]   (optionsManager: firrtl.ExecutionOptionsManager with chisel3.HasChiselExecutionOptions with firrtl.HasFirrtlOptions,dut: () => chisel3.experimental.RawModule)chisel3.ChiselExecutionResult
[error]  cannot be applied to (Array[String], () => dti_bypass_register)
[error]   chisel3.Driver.execute(args, () => new dti_bypass_register)
[error]                  ^
[error] one error found
[error] (Compile / compileIncremental) Compilation failed

What is wrong? How can I fix it? My code is here
`import chisel3._
import chisel3.util._
  class dti_bypass_register extends BlackBox with HasBlackBoxResource {
  val io = IO(new Bundle {
    val clk_DR             = Input (Clock())// Bypass register clock
    val TDI                   = Input (UInt(1.W))// data in
    val bypass_en       = Input (Bool())// enable signal
    val captureDR       = Input (Bool())// captureDR signal

    val TDO_bypass      = Output (UInt(1.W))// Serial data out
  })
    setResource("/dti_bypass_register.v")
}

object dti_bypass_registerDriver extends App {
  chisel3.Driver.execute(args, () => new dti_bypass_register)
}
`

diningyo, 2019-07-30:
Maybe you forgot to wrap your Blackbox module with chisel3.Module class.
The 2nd argument of `chisel3.Driver.execute` must need to pass subclass of RawModule.
You should try to make wrapper module to wrap your Blackbox module class like this:

```scala
class dti_wrapper extends Module {
  val io = IO(new Bundle {
    val clk_DR = Input (Clock())// Bypass register clock
    val TDI = Input (UInt(1.W))// data in
    val bypass_en = Input (Bool())// enable signal
    val captureDR = Input (Bool())// captureDR signal

    val TDO_bypass      = Output (UInt(1.W))// Serial data out
  })

  val mod = Module(new dti_bypass_register)

  io <> mod.io
}
```

Nguyendien, 2019-07-31:
yeb, it is reason why I make this error. I fixed it. Thanks you so much!

edwardcwang, 2019-07-31:
Great to hear it!

closed 2019-07-31 by edwardcwang