chipsalliance/chisel #1145: Blackbox code for a part

Nguyendien, 2019-08-02:
In verilog, we have 
`
module example(
input                               clock;             // Bypass register clock
input                               in;                // data in
input                               enable;          // enable signal
input                               capture;          // capture signal
output          reg                 out;         // data serial out
);

always @(negedge clock or posedge capture)
begin:shift_data_process
  if (capture)
    out <= 1'b0;
  else
  begin
    if (enable)
      out <= in;
  end
end
endmodule
`
Because we can't generate `always @(negedge clock or posedge capture)` to chisel. Instead of sing blacbox resource to blacbox a file, how can I blackbox one line code `always @(negedge clock or posedge capture)`?

chick, 2019-08-11:
You can't do just the one line you need to make the verilog black box contain your whole code snippet. You will need to create inputs and outputs to bring your wires `clock`, `capture`, `out`,  and `in` into your black box and then out again. I can't tell if you want a register in there, or what hardware you need in you black box

Nguyendien, 2019-08-12:
I send you a message on gitter, can you answer to me ?

chick, 2019-08-23:
I believe this is settled. Please re-open with what's unfinished if I'm wrong.

closed 2020-03-16 by chick