chipsalliance/chisel #1170: Incorrect Error Message in Data.connect(...)

ekiwi, 2019-09-03:
In https://github.com/freechipsproject/chisel3/blob/master/chiselFrontend/src/main/scala/chisel3/Data.scala#L377:

```.scala
      requireIsHardware(this, "data to be connected")
      requireIsHardware(that, "data to be connected")
```

The first error message is incorrect. It should be something like `left hand side of the connection`.

mwachs5, 2022-06-28:
Fixed (I believe) by https://github.com/chipsalliance/chisel3/pull/2248

mwachs5, 2022-06-28:
Here is 3.5.3 version:

https://scastie.scala-lang.org/LWzIdu6CSCSh6J94420fPg

which 
```
io.out := io.in
  val foo = inType // This is not hardware since we did not use `Wire(...)` or `Reg(...)`
  foo := io.in // should trigger your new error message about the left-hand side not being hardware
```
produces:
```
chisel3.package$ExpectedHardwareException: data to be connected 'UInt<8>' must be hardware, not a bare Chisel type. Perhaps you forgot to wrap it in Wire(_) or IO(_)?
	at ... ()
```

So is this still the same issue or is this the expected fix for this? It doesn't explicitly say "left hand side of the connection"

ekiwi, 2022-06-28:
> So is this still the same issue or is this the expected fix for this? It doesn't explicitly say "left hand side of the connection"

There is definitely room for improvement. Some points of the top of my head:
- instead of saying `UInt<8>`, it would be cool if we could access the name provided by the plugin so that the message would say `foo`
- `data to be connected` sound unclear to me, alternatives would be `the left hand side of the assignment on line ....` or `the signal foo most be a Wire/Reg or IO, but is just a type (UInt<8>)`

closed 2026-09-21 by jackkoenig