chipsalliance/chisel #1171: Tester sending incorrect signal to verilator when testing RRArbiter with UInt(0.W)

hukangha, 2019-09-05:
**Type of issue**: bug report 

**Impact**:  unknown

**Development Phase**: request | proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
I'm Trying to get if RRArbiter works when discarding the bits signal and only use the valid and chosen signal
so I came up with this testing code
`
class ArbiterTester(c: RRArbiter[UInt] ) extends PeekPokeTester(c) {

  private val abtr = c
  poke(abtr.io.out.ready, 1)
  poke(abtr.io.in(0).valid, 0)
  poke(abtr.io.in(1).valid, 0)
  poke(abtr.io.in(2).valid, 1)
  poke(abtr.io.in(3).valid, 0)
  poke(abtr.io.in(4).valid, 0)
  println(s"out.chosen: ${peek(abtr.io.chosen)}")
  expect(abtr.io.out.valid, 1)
  step(1)
  poke(abtr.io.in(1).valid, 1)
  poke(abtr.io.in(3).valid, 1)
  println(s"out.chosen: ${peek(abtr.io.chosen)}")
  expect(abtr.io.out.valid, 1)
  step(1)
  println(s"out.chosen: ${peek(abtr.io.chosen)}")
  expect(abtr.io.out.valid, 1)
  step(1)
  println(s"out.chosen: ${peek(abtr.io.chosen)}")
  expect(abtr.io.out.valid, 1)
  step(10)

}

class ABTTester extends ChiselFlatSpec {
  // Disable this until we fix isCommandAvailable to swallow stderr along with stdout
  private val backendNames = if(firrtl.FileUtils.isCommandAvailable(Seq("verilator", "--version"))) {
    Array("firrtl", "verilator")
  }
  else {
    Array("firrtl")
  }

  for ( backendName <- backendNames ) {
    "GCD" should s"calculate proper greatest common denominator (with $backendName)" in {
      Driver(() => new RRArbiter(UInt(**0.W**),5), backendName) {
        c => new ArbiterTester(c)
      } should be (true)
    }
  }

}
`

**What is the current behavior?**
the input signal is incorrect shown in the vcd generated by verilator:
[image]

**What is the expected behavior?**
this is the correct signal when setting the type to UInt(1.W):
[image]

as I can get correct verilog code when discarding the bits signal with UInt(0.W),
seems the simulation should work properly.

**Please tell us about your environment:**
version := "3.1.1"

scalaVersion := "2.11.12"

crossScalaVersions := Seq("2.11.12", "2.12.4")

resolvers ++= Seq(
  Resolver.sonatypeRepo("snapshots"),
  Resolver.sonatypeRepo("releases")
)

// Provide a managed dependency on X if -DXVersion="" is supplied on the command line.
val defaultVersions = Map(
  "chisel3" -> "3.1.+",
  "chisel-iotesters" -> "[1.2.5,1.3-SNAPSHOT["
  )

libraryDependencies ++= Seq("chisel3","chisel-iotesters").map {
  dep: String => "edu.berkeley.cs" %% dep % sys.props.getOrElse(dep + "Version", defaultVersions(dep)) }

    - OS: `4.15.0-58-generic ubuntu18.04`

**What is the use case for changing the behavior?**

seldridge, 2019-09-30:
Can you verify if this works is still an issue on the latest release candidate?

seldridge, 2019-09-30:
Thanks for reporting this and thanks for the clear failing test case. 

I *think* this is already fixed, however. I was able to reproduce your failure on all chisel-testers versions including `Seq("1.2.5", "1.2.6", "1.2.7", "1.2.8", "1.2.9", "1.2.10")`, but not on `1.3.0-RC2`.

For more info on the above comment, you can use the following in your `build.sbt` to switch to the `1.3.0` release candidate.

```scala
val defaultVersions = Map(
  "chisel-iotesters" -> "1.3.0-RC2"
  )

libraryDependencies ++= Seq("chisel-iotesters").map {
  dep: String => "edu.berkeley.cs" %% dep % sys.props.getOrElse(dep + "Version", defaultVersions(dep)) }
```

*Note: I'm only pulling in `chisel-iotesters` as a dependency as that will pull in the correct chisel3 dependency.*

johnsbrew, 2019-09-30:
Duplicated issue already answered here: https://github.com/freechipsproject/chisel-testers/issues/258

seldridge, 2019-09-30:
Ah, good catch @johnsbrew! Closing this.

closed 2019-09-30 by seldridge