chipsalliance/chisel #1184: Chisel does not generate error if size argument is not .W in RegInit

Martoni, 2019-09-23:
**Type of issue**: bug report 
**Impact**: no functional change 
**Development Phase**: request

**What is the current behavior?**
If I instantiate a register this way :
```scala
val sizeFrame = 30
val writeFrame = RegInit(0.U(sizeFrame))
```
Chisel compile without problem. But the size of writeFrame register will be 1.

I think it's a chisel bug, because width parameter must be suffixed by '.W'.

**What is the expected behavior?**
 In my opinion, the correct way to instantiate my register is : 
```scala
val sizeFrame = 30
val writeFrame = RegInit(0.U(sizeFrame.W))
```

And the declaration given at the begining should generate an error.

**Please tell us about your environment:**
    - version: `3.1.8`
    - OS: `Linux Debian 4.9.0-9-amd64 #1 SMP Debian 4.9.168-1+deb9u5 (2019-08-11) x86_64 GNU/Linux`

edwardcwang, 2019-09-23:
Hey Fabien, so it turns out this is a long-standing bug that's difficult to prevent (#871) via types as they are currently written. The short answer is that it looks something like the following:

```
// This can be any UInt
val any_UInt = 0.U
// Now extract bit number 30
val bit_30 = any_UInt(30)
```

You'll notice that this is the same as `0.U(30)` :(

At Chisel dev meetings this has come up a few times with the suggestion of linting, or discouraging `0.U(x.W)` etc with no concrete conclusion.

Martoni, 2019-09-23:
Hi Edward, sorry for the duplicate. I didn't find it, but I think it's because it is a closed issue.
That's sad we can not detect this error :(

edwardcwang, 2019-09-24:
I'm not even sure if strict semantics can fix this, unless somehow we introduce a new bit-extract operator? e.g. `myUIntWire.bitExtract(30)`?

schoeberl, 2020-01-18:
As the other issues was closed, I have to continue my ranting herer ;-)

So what is the meaning of
```
0.U(8)
```
except a typo? This results in ```UInt<1>("h00")``` in FIRRTL.

I do not (yet) understand why we cannot deprecate this in one release and then simply have a compile error in a following.

I remember the issue with UInt(n). However, I thought this was cleaned up with Chisel 3 and the new notion of n.U and w.W. I really doubt that there is any (correct) legacy code around that uses c.U(n), with n a Scala Int. I guess if so, it is an error ;-)

Cheers,
Martin

ducky64, 2020-01-18:
From the example in #871 
```scala
val foo = 0.U // declare a UInt whose value is 0
val bar = foo(sz) // extract bit sz from foo
```

Basically, from a Scala operations point of view, it looks like a literal chained with a bit extract, which is legal and apparently in use - and which is why it's so hard to fix.

One solution might be the use of macros (or a linter) to detect and warn on style-based anti-patterns like this. Or we might define a `.U(int)` function that is deprecated / errors and requires users to introduce parens to make chained literal - bit extract operations explicit.

schoeberl, 2020-01-18:
strong +1 to make .U(Int) deprecated/error. What is the argument against it?

PS, I do not really understand your example. What bit length do we expect from 0.U? Why should 0.U(n) help here, when n has almost nothing to do with bit length?

Cheers,
Martin

ducky64, 2020-01-19:
Parse it as (0.U)(8) - create a literal 0 and return bit 8 by bit slicing.

aswaterman, 2020-01-19:
I think it's reasonable to ask people to write `(k.U)(n)` instead of `k.U(n)` to extract a bit from a literal, but unfortunately it is backwards-incompatible, so would need to follow a deprecation schedule or be enabled by a "strict" setting.  I see a small number of places where we (intentionally) extract from literals in this manner in our code.

Martoni, 2020-01-19:
> or be enabled by a "strict" setting.

Is there a plan to add «strict» setting feature in Chisel ?

ducky64, 2020-01-19:
The idea has been thrown around and there appears to be general support for it, but the specifics are still up in the air. See #1163 (implicit truncation, the antifeature that is largely driving strict mode) for the main discussion.

Soya-Onishi, 2021-01-28:
Is there still no conclusion?

I agree with schoeberl.
Bit extraction should only allow only a UInt argument like `15.U(value.U)`.

Fortunately, UInt is allowed to extract BigInt literal from instance by `litOption`, so there is no need to care about extra hardware.

edwardcwang, 2021-01-28:
Note that `value` need not be a hardware type, because it should also be valid to extract a static bit vs a hardware bit (requires muxes).

Martoni, 2021-01-28:
Maybe with a big warning ?

Soya-Onishi, 2021-01-28:
> Note that value need not be a hardware type, because it should also be valid to extract a static bit vs a hardware bit

I do not know what you exactly mean.
However, If what you say means that using value.U cause muxes, I argue that there is no need to care about that  because `litOption` return BigInt type `value`'s value and it is equivalent to static bit extraction.

If you says that there is a reasonable point to use hardware bit extraction (dynamic bit extraction), please tell me what.

edwardcwang, 2021-01-28:
The idea is, say if we have `myWire` which is a UInt, it should be valid to statically extract bit 3 vs to dynamically extract `myWire(myOtherWire)`. They generate different hardware.

Soya-Onishi, 2021-01-28:
I agree that using Int value to extract bit statically is valid, but this can cause erroneous code (i.e. `8.U(3.W)` vs `8.U(3)`).  
I think using literal UInt instead of using Int to extract bit is one of the solutions to prevent above erroneous code.  

`8.U(3.U)` still be static bit extraction although index value type is UInt, and as far as I know generated code is almost equivalent to the case of using Int. Because there is a way to extract bit statically without Int and BigInt, I think these method can be dropped from API eventually.

If you aim to separate between Int and UInt usage to distinguish whether dynamic one or static one, I think that does not make sense because optimization phase will transform into static ones as possible.

jackkoenig, 2021-01-28:
`8.U(3)` is a huge wart that merits fixing. Removing indexing with `Int` would fix it but is a pretty draconian change that would require updating a lot of code (indexing Wires, Regs and the like). There are a couple of other approaches we could do:
* I tried to specifically deprecate (for removal) indexing a literal immediately. It's a little tricky but it could maybe work: https://github.com/chipsalliance/chisel3/compare/deprecate-literal-bit-extract
* @albert-magyar did a proof of concept of making this a linter error using ScalaFix: https://github.com/chipsalliance/chisel3/pull/1449

Soya-Onishi, 2021-01-29:
I also thought about using scalafix, but first one also looks good for me.

Another solution is throwing an error when meeting below conditions

- literal UInt (`litOption` returns `Some`)
- unknown width
- out-of-bounds from least width to represent literal value

In the case of typo like `8.U(4)` instead of `8.U(4.W)`, this meets above condition. Current implementation also require enough length to represent literals (i.e. 8.U(3.W) causes an error), so it is able to reject the pattern like appropriate width and typo although inappropriate width and typo pattern (i.e. 8.U(3)) still cause no errors.

Above solution preserves current API interfaces, but, as discussed in #857, there are use cases that uses out-of-bounds bit extraction.

#867 also proposes adding new API `padExtract` to solve out-of-bounds bit extraction.
I agree to add `padExtract` as a one of the solutions to resolve this issue(I also think it looks like Map's `getOrElse` method although default value is implicit one instead of explict one) .

Soya-Onishi, 2021-01-29:
Sorry for the misunderstanding about literal width.

literal width is automatically calculated, so the case of unknown value with literal value does not happen.
That's why there is only need the out-of-bounds check.

mwachs5, 2022-06-02:
Closed by #2534

closed 2022-06-02 by mwachs5