chipsalliance/chisel #1185: Enum(n) error better message

Martoni, 2019-09-23:
**Type of issue**: feature request
**Impact**: no functional change
**Development Phase**: proposal
**What is the current behavior?**

When we declare Enum list with wrong size in parameter:
```scala 
  // example of Enum list with wrong size
  val sheadaddr::swriteframe::sreaddata::sidle::Nil = Enum(6)
```

We get a really bad scala stack error :
```
[info] [0,001] Elaborating design...
[error] (run-main-0) scala.MatchError: List(chisel3.core.UInt@da, chisel3.core.UInt@db, chisel3.core.UInt@dc, chisel3.core.UInt@dd, chisel3.core.UInt@de, chisel3.core.UInt@df) (of class scala.collection.immutable.$colon$colon)
[error] scala.MatchError: List(chisel3.core.UInt@da, chisel3.core.UInt@db, chisel3.core.UInt@dc, chisel3.core.UInt@dd, chisel3.core.UInt@de, chisel3.core.UInt@df) (of class scala.collection.immutable.$colon$colon)
[error] 	at mdio.Mdio.<init>(mdio.scala:107)
[error] 	at mdio.Mdio$$anonfun$17.apply(mdio.scala:159)
[error] 	at mdio.Mdio$$anonfun$17.apply(mdio.scala:159)
[error] 	at chisel3.core.Module$.do_apply(Module.scala:49)
[error] 	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:93)
[error] 	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:93)
[error] 	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:297)
[error] 	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:295)
[error] 	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error] 	at chisel3.internal.Builder$.build(Builder.scala:295)
[error] 	at chisel3.Driver$.elaborate(Driver.scala:93)
[error] 	at chisel3.Driver$.execute(Driver.scala:140)
[error] 	at chisel3.Driver$.execute(Driver.scala:202)
[error] 	at mdio.Mdio$.delayedEndpoint$mdio$Mdio$1(mdio.scala:159)
[error] 	at mdio.Mdio$delayedInit$body.apply(mdio.scala:153)
[error] 	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
[error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
[error] 	at scala.collection.immutable.List.foreach(List.scala:381)
[error] 	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
[error] 	at scala.App$class.main(App.scala:76)
[error] 	at mdio.Mdio$.main(mdio.scala:153)
[error] 	at mdio.Mdio.main(mdio.scala)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error] 	at java.lang.reflect.Method.invoke(Method.java:498)
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 3 s, completed 23 sept. 2019 15:30:11
```

**What is the expected behavior?**

Is it possible to have better error message for it like "wrong Enum size" ? 
Or better : is it possible to declare Enum with automatic size ?

**Please tell us about your environment:**

    - version: `3.1.8`
    - OS: `Linux Debian Linux fabien 4.9.0-9-amd64 #1 SMP Debian 4.9.168-1+deb9u5 (2019-08-11) x86_64 GNU/Linux
`

seldridge, 2019-09-30:
What's difficult about this is that the basic `Enum(n: Int)` returns a `List[UInt]` and the assignment of this to `sheadaddr`, etc. is done via the extractor pattern. I.e., this is really equivalent to the following:

```scala
val List(a, b, c) = List(1, 2, 3)
/* Or, using list notation: */
val aa :: bb :: cc :: Nil = List(4, 5, 6)
```

Using different lengths gives you the same error:
```
scala> val List(a, b, c, d) = List(1, 2, 3)
scala.MatchError: List(1, 2, 3) (of class scala.collection.immutable.$colon$colon)
  ... 33 elided

scala> val List(a, b) = List(1, 2, 3)
scala.MatchError: List(1, 2, 3) (of class scala.collection.immutable.$colon$colon)
  ... 33 elided
```

Two ideas: (1) take a look at the new `chisel3.experimental.ChiselEnum`. That is better overall. Also, (2) you could do something like the following jank:

```scala
val enums = Seq("sheadaddr", "swriteframe", "sreaddata") match { case x => x.zip(Enum(x.size)).toMap }
/* You can then do a janky, **stringly typed** dereference, e.g.: */
val x = enums("sheadaddr")
```

Alternatively, there may be a way that `Enum` could return a special sequence with `unapplySeq` overridden that allows for a better error message here.

Martoni, 2019-09-30:
Thanks, I will try ChiselEnum.

jackkoenig, 2023-05-09:
Solution is to use `ChiselEnum`.