chipsalliance/chisel #1192: Unexpected behavior when composing Flipped and Vec

spaceotter, 2019-09-25:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**What is the current behavior?**
```scala
val in = Flipped(Vec(inChannels, Decoupled(Bool()) ))
```
has different behavior than:
```scala
val in = Vec(inChannels, Flipped(Decoupled(Bool())) ))
```
When doing something like this:
```
for(i <- 0 until inChannels) { io.in(i).ready := false.B }
...
io.in(io.chan).ready := true.B
```
The second type definition and second assignment produces an exception during elaboration:
```
[error] chisel3.internal.ChiselException: Connection between sink (Bool(IO in unelaborated MyModule)) and source (Bool(true)) failed @: Sink is unwriteable by current module.
```
Changing to the first type clears the exception.

**What is the expected behavior?**
The two forms should be equivalent.

**Please tell us about your environment:**
chisel: 3.2.0-RC2
scala: 2.12.8

seldridge, 2019-09-25:
This is just reiterating the issue, but with a more verbose, self-contained test case.

```scala
import chisel3._
import chisel3.util.{Decoupled, log2Up}
import chisel3.stage.{ChiselStage, ChiselGeneratorAnnotation}

class Foo(n: Int) extends RawModule {
  val inFV = IO(Flipped(Vec(n, Decoupled(Bool()))))
  val inVF = IO(Vec(n, Flipped(Decoupled(Bool()))))
  val sel = IO(Input(UInt(log2Up(n).W)))

  (0 until n).foreach{ case i =>
    inFV(i).ready := false.B
    inVF(i).ready := false.B
  }

  inFV(sel).ready := true.B
  inVF(sel).ready := true.B /* This line is the problem */
}

(new ChiselStage)
  .execute(Array("-X", "verilog"), Seq(ChiselGeneratorAnnotation(() => new Foo(4))))
```

My guess would be to look at either `cloneTypeFull` or with the `do_apply(p: UInt)` method. This works fine with the `apply(idx: Int)` method.

jackkoenig, 2019-09-25:
I feel like I'm taking crazy pills, this is seems similar to https://github.com/freechipsproject/chisel3/issues/1063 which was similar to https://github.com/freechipsproject/chisel3/pull/945

jackkoenig, 2019-09-27:
I've been investigating but am out of time today so I'm writing down what I've found.

It appears to be due to https://github.com/freechipsproject/chisel3/blob/9746aa39d8b005dee8a523ad85ce2059f8f1c842/chiselFrontend/src/main/scala/chisel3/Aggregate.scala#L223 conflicting with https://github.com/freechipsproject/chisel3/blob/9746aa39d8b005dee8a523ad85ce2059f8f1c842/chiselFrontend/src/main/scala/chisel3/Aggregate.scala#L159. The reconstruction might work if we stop setting the direction directly from the child (an optimization) or if we can differentiate between `ActualDirection.Bidirectional(ActualDirection.Default)` and `ActualDirection.Bidirectional(ActualDirection.Flipped)` properly.

These tests can be enhanced to check for this: https://github.com/freechipsproject/chisel3/blob/9746aa39d8b005dee8a523ad85ce2059f8f1c842/src/test/scala/chiselTests/Direction.scala#L201
We should add an index and check the directions of the result of dynamic indexing.

Another test case to consider (my attempts at fixing worked but failed this)
```scala
IO(Flipped(new Bundle { val x = Flipped(Vec(2, Flipped(new MyBundle))) }))
```

commit 80c1cd21 (2019-09-28): Fix direction of dynamic index in complex Vec

Dynamically indexing a Vec of Flipped bidirectional Bundles would get
the wrong directions on the elements of the Bundles

Fixes #1192
PR #1196 (2019-10-08): Fix direction of dynamic index in complex Vec
Dynamically indexing a Vec of Flipped bidirectional Bundles would get
the wrong directions on the elements of the Bundles

Fixes #1192

See https://github.com/freechipsproject/chisel3/issues/1192#issuecomment-536018644 for insight into the issue. Basically, we were copying the direction from the `sample_element` to the `Vec` itself as an optimization, but if you have a Flipped Bundle as the element this could result in double flipping. Instead, we want to maintain a different direction for the Vec itself so that we can differentiate between complex [... cut]
commit fafd984a (2019-10-08): Fix direction of dynamic index in complex Vec (#1196)

Dynamically indexing a Vec of Flipped bidirectional Bundles would get
the wrong directions on the elements of the Bundles

Fixes #1192
closed 2019-10-08 by seldridge