chipsalliance/chisel #1195: Test Failed with chisel-testers

medabme, 2019-09-28:
I got a problem when running tests whit chisel-testers. 
I used ComplexMemoryTester program from [https://github.com/freechipsproject/chisel-testers/blob/master/src/test/scala/examples/ComplexMemoryLoadingSpec.scala](url) to load a file from memory. This is the code:

`
class MemoryShape extends Bundle {
  val a = UInt(65536.W)

}

class HasComplexMemory(memoryDepth: Int) extends Module {
  val io = IO(new Bundle {
    val address = Input(UInt(log2Ceil(memoryDepth).W))
    val value   = Output(new MemoryShape)
    val Buffer  = Output(Vec(100, UInt(32.W)))
  })

   val H  = new ArrayBuffer[UInt]()
    for ( k <- 0 until 100) {
      H += 0.asUInt(32.W)    }
    val memory = Mem(memoryDepth, new MemoryShape)
   loadMemoryFromFile(memory, "test_run_dir/complex_mem_test/cameraman")
    io.value := memory(io.address)
    io.Buffer := H
}

class HasComplexMemoryTester(c: HasComplexMemory) extends PeekPokeTester(c) {
  for(addr <- 0 until 65536) {
    poke(c.io.address, addr)
    step(1)
    println(f"peek from $addr ${peek(c.io.value.a)}%x")
    poke(c.io.Buffer(addr), addr )
    println(peek(c.io.Buffer(addr)).toString)

  }
  println("Done compiling")
}

class ComplexMemoryLoadingSpec extends  FreeSpec with Matchers {
  "memory loading should be possible with complex memories" - {

    val targetDirName = "test_run_dir/complex_mem_test"
    FileUtils.makeDirectory(targetDirName)

    val path1 = Paths.get(targetDirName + "/cameraman_a")
    Files.copy(getClass.getResourceAsStream("/cameraman.txt"), path1, REPLACE_EXISTING)

    "should work with verilator" in {
      iotesters.Driver.execute(
        args = Array("--backend-name", "verilator", "--target-dir", targetDirName, "--top-name", "complex_mem_test"),
        dut = () => new HasComplexMemory(memoryDepth = 65536)
      ) { c =>
        new HasComplexMemoryTester(c)
      } should be(true)
    }
  }
}
` 

After running the test some errors appear, the test failed and the simulation stops.
This is the complete output:

`
[info] Done compiling.
[info] [0.008] Elaborating design...
[info] [0.455] Done elaborating.
Total FIRRTL Compile Time: 1975.7 ms
cd /home/med/workspace/ChiselProjects/convolution/test_run_dir/complex_mem_test && verilator --cc HasComplexMemory.v --assert -Wno-fatal -Wno-WIDTH -Wno-STMTDLY -O1 --top-module HasComplexMemory +define+TOP_TYPE=VHasComplexMemory +define+PRINTF_COND=!HasComplexMemory.reset +define+STOP_COND=!HasComplexMemory.reset -CFLAGS "-Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VHasComplexMemory -DVL_USER_FINISH -include VHasComplexMemory.h" -Mdir /home/med/workspace/ChiselProjects/convolution/test_run_dir/complex_mem_test -f /home/med/workspace/ChiselProjects/convolution/test_run_dir/complex_mem_test/firrtl_black_box_resource_files.f --exe /home/med/workspace/ChiselProjects/convolution/test_run_dir/complex_mem_test/HasComplexMemory-harness.cpp --trace
make: Entering directory '/home/med/workspace/ChiselProjects/convolution/test_run_dir/complex_mem_test'
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VHasComplexMemory -DVL_USER_FINISH -include VHasComplexMemory.h   -c -o HasComplexMemory-harness.o /home/med/workspace/ChiselProjects/convolution/test_run_dir/complex_mem_test/HasComplexMemory-harness.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VHasComplexMemory -DVL_USER_FINISH -include VHasComplexMemory.h   -c -o verilated.o /usr/local/share/verilator/include/verilated.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VHasComplexMemory -DVL_USER_FINISH -include VHasComplexMemory.h   -c -o verilated_vcd_c.o /usr/local/share/verilator/include/verilated_vcd_c.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VHasComplexMemory.cpp VHasComplexMemory___024unit.cpp > VHasComplexMemory__ALLcls.cpp
/usr/bin/perl /usr/local/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VHasComplexMemory__Trace.cpp VHasComplexMemory__Syms.cpp VHasComplexMemory__Trace__Slow.cpp > VHasComplexMemory__ALLsup.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VHasComplexMemory -DVL_USER_FINISH -include VHasComplexMemory.h   -c -o VHasComplexMemory__ALLsup.o VHasComplexMemory__ALLsup.cpp
g++  -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -Wno-undefined-bool-conversion -O1 -DTOP_TYPE=VHasComplexMemory -DVL_USER_FINISH -include VHasComplexMemory.h   -c -o VHasComplexMemory__ALLcls.o VHasComplexMemory__ALLcls.cpp
ar: creating VHasComplexMemory__ALL.a
      Archiving VHasComplexMemory__ALL.a ...
ar r VHasComplexMemory__ALL.a VHasComplexMemory__ALLcls.o VHasComplexMemory__ALLsup.o
ranlib VHasComplexMemory__ALL.a
g++    HasComplexMemory-harness.o verilated.o verilated_vcd_c.o VHasComplexMemory__ALL.a    -o VHasComplexMemory -lm -lstdc++ 
make: Leaving directory '/home/med/workspace/ChiselProjects/convolution/test_run_dir/complex_mem_test'
sim start on Dell at Sat Sep 28 01:53:11 2019
inChannelName: 00017728.in
outChannelName: 00017728.out
cmdChannelName: 00017728.cmd
STARTING test_run_dir/complex_mem_test/VHasComplexMemory
Enabling waves..%Error: test_run_dir/complex_mem_test/cameraman_a:1: $readmem file syntax error
Aborting...
[info] ComplexMemoryLoadingSpec:
[info] memory loading should be possible with complex memories
[info] - should work with verilator *** FAILED ***
[info]   chisel3.iotesters.TestApplicationException: test application exit - exit code 134
[info]   at chisel3.iotesters.SimApiInterface.throwExceptionIfDead(SimApiInterface.scala:98)
[info]   at chisel3.iotesters.SimApiInterface.chisel3$iotesters$SimApiInterface$$mwhile(SimApiInterface.scala:107)
[info]   at chisel3.iotesters.SimApiInterface$$anonfun$start$3.apply$mcVI$sp(SimApiInterface.scala:268)
[info]   at scala.collection.immutable.Range.foreach$mVc$sp(Range.scala:160)
[info]   at chisel3.iotesters.SimApiInterface.start(SimApiInterface.scala:266)
[info]   at chisel3.iotesters.SimApiInterface.<init>(SimApiInterface.scala:325)
[info]   at chisel3.iotesters.VerilatorBackend.<init>(VerilatorBackend.scala:295)
[info]   at chisel3.iotesters.setupVerilatorBackend$.apply(VerilatorBackend.scala:284)
[info]   at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply$mcZ$sp(Driver.scala:56)
[info]   at chisel3.iotesters.Driver$$anonfun$execute$1$$anonfun$apply$mcZ$sp$1.apply(Driver.scala:39)
[info]   ...
[info] ScalaTest
[info] Run completed in 1 minute, 53 seconds.
[info] Total number of tests run: 1
[info] Suites: completed 1, aborted 0
[info] Tests: succeeded 0, failed 1, canceled 0, ignored 0, pending 0
[info] *** 1 TEST FAILED ***
[error] Failed: Total 1, Failed 1, Errors 0, Passed 0
[error] Failed tests:
[error] 	util.ComplexMemoryLoadingSpec
[error] (Test / testOnly) sbt.TestsFailedException: Tests unsuccessful
`  
Can someone explain the error please or help me to fix it.

seldridge, 2019-09-28:
The specific error [`"$readmem file syntax error"`](https://github.com/verilator/verilator/blob/7e54ff1b375a5c8c646a08e12f2b27bd1fd67e85/include/verilated.cpp#L1607) is a Verilator error indicating, in my understanding, that the file pointed to by `$readmemh` or `$readmemb` has a bad format. 

If you are using `loadMemoryFromFile` with the hex option (the default), this will map to `$readmemh`. The file should then look like something like:

```
1
2
3
dead
beef
```

medabme, 2019-09-28:
Thanks for answering me!
I didn't use the same files in the example. I used in this program my own file and it is a .txt file (it contains binary numbers 0 and 1), is .txt files cause problems with Verilator? by the way i used w a .bin file before and it causes the same error.

chick, 2019-09-28:
I think you are close to getting this to work. There are two problems.
Problem1: The io.Buffer. This is not related to memory.

It looks like you would like to have 100 32bit registers that you can write to and read from.

As declared Buffer in your code is Output, even though testers will let you poke it this will result in a run-time error. The way you are initializing is also problematic. I think it would be better (although I am not certain of your use-case) to do something like the following which allocates a bank of registers to hold the values you poke in.
The ArrayBuffer you initialize is pointing at literal values, which cannot be changed.
The `H := ...` you had before does not do what you would like,it compiles but will also cause run-time problem.
```scala
class HCM(memoryDepth: Int) extends Module {
  val io = IO(new Bundle {
    val address = Input(UInt(log2Ceil(memoryDepth).W))
    val value   = Output(new MShape)
    val inBuffer  = Input(Vec(100, UInt(32.W)))
    val outBuffer  = Output(Vec(100, UInt(32.W)))
  })

  io.inBuffer.zip(io.outBuffer).foreach { case (in, out) =>
    val reg = RegInit(0.U(32.W))
    reg := in
    out := reg
  }

  val memory = Mem(memoryDepth, new MShape)
  loadMemoryFromFile(memory, "test_run_dir/cmt/cameraman")
  io.value := memory(io.address)
}
```
and your tester code will look like
```scala
class HCMTester(c: HCM) extends PeekPokeTester(c) {
  for(addr <- 0 until 65536) {
    poke(c.io.address, addr)
    step(1)
    println(f"peek from $addr ${peek(c.io.value.a)}%x")
    poke(c.io.inBuffer(addr), addr )
    println(peek(c.io.outBuffer(addr)).toString)

  }
  println("Done compiling")
}
```

Problems with memory will be addressed in the next comment

chick, 2019-09-29:
Problem 2: Getting your memory initialized.
I would suggest that you remove or comment out the Buffer stuff I talked about above until you have worked out the memory loading, just to keep it simpler to figure out what's going on.

The error you are getting makes it seem likely that you have things in the right place, i.e.  relative to your repo's root directory it should look like this.
```
> find src/test/resources
src/test/resources/
src/test/resources//cameraman_a.txt
```
so the problem is likely to be with the contents of your file.
@seldridge gives a good example of the what the file should look like internally.

Can you share the contents of the file or at least the first few text lines of it?

Also: The width of your memory is very large. This makes testing slower, especially when running with verilator. Did you really want your memory to be 64k bits wide?

Cheers, I hope this helps. Let me know how it turns out.

medabme, 2019-09-29:
@chick thanks for your time. Let me explain the purpose of this program; 
1) I m trying to find a way to load a file into memory (cameraman.txt)
2) I m trying to read values from the file and write them in the `io.Buffer`
3) It prints the 100 values written in the declared register (just to verify if the copy operation really works)

I used 65k bits memory, simply because my file sized 65k bits. 
Here is the contents of the file(i copied few lines)

`00000100 00000000 00000100 00100000 11010010 00000000 00000000 11111111 11010100 00000000 00000000 11111111 11010101 00000000 00000000 11111111 11010101 00000000 00000000 11111111 11010011 00000000 00000000 11111111 11010011 00000000 00000000 11111111 11010101 00000000`

It might be a wrong method to do the job so don't hesitate to suggest modifications and feel free to share your ideas.

chick, 2019-10-01:
1. I think you are almost setup to load the memory at simulation startup
    - I think the only thing causing you problems is the format of your load memory file `cameraman.txt`.
        -   @seldridge provided an example, it must be one value per line.
        - the all the values in the file must be the same format, the default is hex
        - if you want binary, there is a 3rd argument to `loadMemoryFromFile` that you must specify
            - it looks like the file you have above is in binary and is not correctly formatted with 1 value per line.
        - it appears your memory
        - See [this article](https://timetoexplore.net/blog/initialize-memory-in-verilog) or others on the details of the file format.
2. The way this works is the memory is loaded at simulation startup, and that can happen only once as far as I know.
    - You can access the memory for reading the way I did in the HCM example above
    - if you want to write new values into memory you must added explicit io fields that will do that
      - this usually requires separate address, data and enable fields that the module can use for writing to the memory
3. The HCMTester example above shows how to read the first 100 numbers from your file. You can easily change it to read more.

> HCM includes a version of the buffers you have in your original post, I am not sure of their intent. I would suggest you first get the memory load working and get the module to support reading and writing from that memory. Then introduce the buffers if you still need them

I hope this is enough to get you on the right track.

chick, 2019-12-18:
Please re-open if more info required

closed 2019-12-18 by chick