chipsalliance/chisel #1197: why the code of statemachine generate if-else statement instead of case statement with Verilog

qgzln, 2019-09-29:
there is no case in the generated Verilog statement, I usually use the case when writing the state machine with Verilog. So, chisel3 still can not generate Verilog code with case？

seldridge, 2019-09-30:
That's correct, Verilog code generated from `when`/`.elsewhen`/`.otherwise` or `switch`/`is` will map to Verilog `if`/`else` blocks. Fortunately, from the perspective of synthesis tools, this *shouldn't* matter in terms of QOR: the tooling should optimize these accordingly.

However, this is clearly sub-optimal in terms of readability of generated Verilog.

The addition of `chisel3.experimental.ChiselEnum` provides both (1) better support for enumerated types and (2) annotates enumerated Chisel types. With both of these supported, there's been a long-standing idea to write an FSM library that would generate more readable state-machines in the output Verilog.

Summarily, this is a cool idea that's looking for a developer to drive it.

schoeberl, 2019-09-30:
+1 on this. I stumbled over a performance issue that the logic of the switch statement can fit into one LUT per bit in an FPGA, but the prioritized selection chain resulted in 2 LUTs per bit. This doubles the hardware size and reduced the fmax compared to a pure Verilog/VHDL implementation of the same circuit.

chick, 2019-12-18:
This seems to be decently answered. Please re-open if more information is needed here.

closed 2019-12-18 by chick

seldridge, 2019-12-19:
I think discussion around this has moved to: https://github.com/freechipsproject/chisel3/issues/1198. 

FIRRTL has merged support to generate `else if` generation via https://github.com/freechipsproject/firrtl/pull/1204. This is available in FIRRTL 1.2.1 (which is used by Chisel 3.2.1).