chipsalliance/chisel #1198: [rfc] Case/If Else Statement Emission

seldridge, 2019-10-01:
Currently, Verilog emission of `when`/`.elsewhen`/`.otherwise` or `switch`/`is` results in a sequence of nested `if`/`else` in the resulting Verilog. While synthesis tools may be perfectly fine with this structure, human users are not fine with this seemingly unintuitive behavior.

This proposal concerns a push to improve this generation by supporting emission of Verilog `case` statements. As a corollary, this proposal *may* naturally also cover generation of `else if` statements.

Roughly, there are two approaches to doing this: (1) a *heuristic* based approach and (2) a library based approach.

Consider the following example:

```scala
val outSwitch = IO(Output(Bool()))
outSwitch := {
  val tmp = RegInit(false.B)
  switch (in) {
    is(0.U) { tmp := false.B }
    is(1.U) { tmp := false.B }
    is(2.U) { tmp := false.B }
    is(3.U) { tmp := true.B }
  }
  tmp
}
```

Producing the following Low FIRRTL:

```
    reg _T : UInt<1>, clock with :
      reset => (UInt<1>("h0"), _T) @[<console> 21:22]
    node _T_1 = eq(UInt<2>("h0"), in) @[Conditional.scala 37:30]
    node _T_2 = eq(UInt<2>("h1"), in) @[Conditional.scala 37:30]
    node _T_3 = eq(UInt<2>("h2"), in) @[Conditional.scala 37:30]
    node _T_4 = eq(UInt<2>("h3"), in) @[Conditional.scala 37:30]
    node _GEN_0 = or(_T_4, _T) @[Conditional.scala 39:67]
    node _GEN_1 = mux(_T_3, UInt<1>("h0"), _GEN_0) @[Conditional.scala 39:67]
    node _GEN_2 = mux(_T_2, UInt<1>("h0"), _GEN_1) @[Conditional.scala 39:67]
    node _GEN_3 = mux(_T_1, UInt<1>("h0"), _GEN_2) @[Conditional.scala 40:58]
```

## Heuristics

The above example has the following properties:

1. The failure condition of each `mux` is a reference to another `mux` (except in the default)
2. The condition for each `mux` is an `eq` involving a constant and a reference (`foo`)
3. Each reference (`foo`) is the same
4. Each constant is mutually exclusive with all other constants

Observing these four properties (which are simple to check), it seems natural that these muxes can be grouped into one `case` statement. 

A concrete way to do this would be to:

1. Create an annotation storing this information
2. Modify the `VerilogEmitter` to use this annotation during emission

Alternatively, this could be done with a `CombineMuxes` transform that replaces references to the failure condition with the actual `Mux`. It's then relatively trivial for the emitter to verify that these conditions continue to hold when emitting.

## Library

In this approach, there is no heuristic and we rely on a new Chisel library (backend by annotations) to guide the emission.

This is potentially more complicated as FIRRTL has no underlying primitive to represent case/switch, only `when`. Here, the above guarantees would have to be checked by an explicit transform or by the emitter to guarantee that no modifications have happened that would result in an improper case.

## QOR

The QOR of the emitted Verilog with respect to FPGA/ASIC toolchains with this proposed modification is unknown at this time.

### Example

Consider the following Chisel code that implements an AND gate using both `switch` and `when`:

```scala
import chisel3._
import chisel3.util.{switch, is, SwitchContext}
import chisel3.stage.{ChiselGeneratorAnnotation, ChiselStage}

class Foo extends MultiIOModule {
  val in = IO(Input(UInt(2.W)))
  val outSwitch = IO(Output(Bool()))
  val outWhen = IO(Output(Bool()))

  outSwitch := {
    val tmp = RegInit(false.B)
    switch (in) {
      is(0.U) { tmp := false.B }
      is(1.U) { tmp := false.B }
      is(2.U) { tmp := false.B }
      is(3.U) { tmp := true.B }
    }
    tmp
  }

  outWhen := {
    val tmp = RegInit(false.B)
    when(in === 0.U)        { tmp := false.B }
      .elsewhen(in === 1.U) { tmp := false.B }
      .elsewhen(in === 2.U) { tmp := false.B }
      .otherwise            { tmp := true.B  }
    tmp
  }
}

(new ChiselStage)
  .execute(Array("-X", "verilog"), Seq(ChiselGeneratorAnnotation(() => new Foo)))
```

This produces the following FIRRTL:

```
;buildInfoPackage: chisel3, version: 3.2.0-RC2, scalaVersion: 2.12.10, sbtVersion: 1.2.7
circuit readiwiwFoo : 
  module readiwiwFoo : 
    input clock : Clock
    input reset : UInt<1>
    input in : UInt<1>
    output outSwitch : UInt<1>
    output outWhen : UInt<1>
    
    reg _T : UInt<1>, clock with : (reset => (reset, UInt<1>("h00"))) @[<pastie> 21:22]
    node _T_1 = eq(UInt<1>("h00"), in) @[Conditional.scala 37:30]
    when _T_1 : @[Conditional.scala 40:58]
      _T <= UInt<1>("h00") @[<pastie> 23:21]
      skip @[Conditional.scala 40:58]
    else : @[Conditional.scala 39:67]
      node _T_2 = eq(UInt<1>("h01"), in) @[Conditional.scala 37:30]
      when _T_2 : @[Conditional.scala 39:67]
        _T <= UInt<1>("h00") @[<pastie> 24:21]
        skip @[Conditional.scala 39:67]
      else : @[Conditional.scala 39:67]
        node _T_3 = eq(UInt<2>("h02"), in) @[Conditional.scala 37:30]
        when _T_3 : @[Conditional.scala 39:67]
          _T <= UInt<1>("h00") @[<pastie> 25:21]
          skip @[Conditional.scala 39:67]
        else : @[Conditional.scala 39:67]
          node _T_4 = eq(UInt<2>("h03"), in) @[Conditional.scala 37:30]
          when _T_4 : @[Conditional.scala 39:67]
            _T <= UInt<1>("h01") @[<pastie> 26:21]
            skip @[Conditional.scala 39:67]
    outSwitch <= _T @[<pastie> 20:13]
    reg _T_5 : UInt<1>, clock with : (reset => (reset, UInt<1>("h00"))) @[<pastie> 32:22]
    node _T_6 = eq(in, UInt<1>("h00")) @[<pastie> 33:13]
    when _T_6 : @[<pastie> 33:29]
      _T_5 <= UInt<1>("h00") @[<pastie> 33:35]
      skip @[<pastie> 33:29]
    else : @[<pastie> 34:29]
      node _T_7 = eq(in, UInt<1>("h01")) @[<pastie> 34:20]
      when _T_7 : @[<pastie> 34:29]
        _T_5 <= UInt<1>("h00") @[<pastie> 34:35]
        skip @[<pastie> 34:29]
      else : @[<pastie> 35:29]
        node _T_8 = eq(in, UInt<2>("h02")) @[<pastie> 35:20]
        when _T_8 : @[<pastie> 35:29]
          _T_5 <= UInt<1>("h00") @[<pastie> 35:35]
          skip @[<pastie> 35:29]
        else : @[<pastie> 36:29]
          _T_5 <= UInt<1>("h01") @[<pastie> 36:35]
          skip @[<pastie> 36:29]
    outWhen <= _T_5 @[<pastie> 31:11]
```

And the following Verilog:

```verilog
module readiwiwFoo(
  input   clock,
  input   reset,
  input   in,
  output  outSwitch,
  output  outWhen
);
  reg  _T; // @[<pastie> 21:22]
  reg [31:0] _RAND_0;
  wire  _T_1; // @[Conditional.scala 37:30]
  wire [1:0] _GEN_7; // @[Conditional.scala 37:30]
  wire  _T_3; // @[Conditional.scala 37:30]
  wire  _T_4; // @[Conditional.scala 37:30]
  wire  _GEN_0; // @[Conditional.scala 39:67]
  reg  _T_5; // @[<pastie> 32:22]
  reg [31:0] _RAND_1;
  wire  _T_6; // @[<pastie> 33:13]
  wire  _T_8; // @[<pastie> 35:20]
  assign _T_1 = 1'h0 == in; // @[Conditional.scala 37:30]
  assign _GEN_7 = {{1'd0}, in}; // @[Conditional.scala 37:30]
  assign _T_3 = 2'h2 == _GEN_7; // @[Conditional.scala 37:30]
  assign _T_4 = 2'h3 == _GEN_7; // @[Conditional.scala 37:30]
  assign _GEN_0 = _T_4 | _T; // @[Conditional.scala 39:67]
  assign _T_6 = in == 1'h0; // @[<pastie> 33:13]
  assign _T_8 = _GEN_7 == 2'h2; // @[<pastie> 35:20]
  assign outSwitch = _T; // @[<pastie> 20:13]
  assign outWhen = _T_5; // @[<pastie> 31:11]
  always @(posedge clock) begin
    if (reset) begin
      _T <= 1'h0;
    end else begin
      if (_T_1) begin
        _T <= 1'h0;
      end else begin
        if (in) begin
          _T <= 1'h0;
        end else begin
          if (_T_3) begin
            _T <= 1'h0;
          end else begin
            _T <= _GEN_0;
          end
        end
      end
    end
    if (reset) begin
      _T_5 <= 1'h0;
    end else begin
      if (_T_6) begin
        _T_5 <= 1'h0;
      end else begin
        if (in) begin
          _T_5 <= 1'h0;
        end else begin
          if (_T_8) begin
            _T_5 <= 1'h0;
          end else begin
            _T_5 <= 1'h1;
          end
        end
      end
    end
  end
endmodule

```

**Type of issue**: feature request

**Impact**: API modification

**Development Phase**: request | proposal

**Other information**
 

**What is the use case for changing the behavior?**

Better Verilog generation.

derekpappas, 2020-01-28:
Consider the following example:

```
val outSwitch = IO(Output(Bool()))
outSwitch := {
  val tmp = RegInit(false.B)
  switch (in) {
    is(0.U) { tmp := false.B }
    is(1.U) { tmp := false.B }
    is(2.U) { tmp := false.B }
    is(3.U) { tmp := true.B }
  }
  tmp
}
```

Can produce the following FIRRTL:

```
    node outSwitch = Switch(Wref("in"),
    List(
    CaseSelect(eq(UInt<2>("h00"), 0)
    CaseSelect(eq(UInt<2>("h1" ), 0)
    CaseSelect(eq(UInt<2>("h2" ), 0)
    CaseSelect(eq(UInt<2>("h3" ), 1)
    )
    )
```

Which can the produce a Verilog cas statement

We will need to define the properties that the selector has:
one hot 
all defines (e.g a 2 bit selector is fully defined if all 4 cases are defined - 00, 01, 10, 11)
has default

Case selctors have don't care's
- how are they defined in Chisel

seldridge, 2020-01-28:
I had not considered implementing this as a new FIRRTL IR construct. Are you thinking of this as something that is in the FIRRTL spec or something that is added only by a Verilog emitter-internal transform?

I had started to implement a different approach on the `case` branch (https://github.com/freechipsproject/firrtl/compare/case), but haven't had time to get back to it. The approach I was playing with is similar to how the new `else-if` emission works (https://github.com/freechipsproject/firrtl/pull/1204): `case` emission would happen if the sequence of `false` conditions in a `mux` tree were all equalities with the same expression and same constant. A default would then just be the last `false` condition. 

This could then be implemented either inside the emitter or with a transform that runs to generate `CaseSelect` blocks.

This proved a little annoying to code up inside the emitter and I was hoping that a clean solution like the `else-if` PR  was possible. Perhaps that lends credence to an internal transform.

derekpappas, 2020-01-29:
> the new else-if emission works (freechipsproject/firrtl#1204

Yes, I am proposing a new FIRRTL IR Case and CaseSelect construct.

I will look at your PR.

derekpappas, 2020-01-29:
Please send me a link to the Chisel code that reads the chisel switch
statement.

seldridge, 2020-01-29:
I'd take a look at [`Conditional.scala`](https://github.com/freechipsproject/chisel3/blob/master/src/main/scala/chisel3/util/Conditional.scala#L37). `switch`/`is` are simply wrappers around `when`/`elsewhen`.

Note: I'd be more in favor of not adding new IR nodes with equivalent functionality to FIRRTL's `when`/`else` in the spirit of keeping the IR as simple as possible, but it's probably better for @azidar to weigh in there.

sequencer, 2020-01-29:
I think 
```
tmp := MuxLookup(in, false.B,
  Array(
    0.U -> false.B,
    1.U -> false.B,
    2.U -> false.B,
    3.U -> true.B
  )
)
```
should be enough, we only need add `switch` mux to `Emitter`.