chipsalliance/chisel #1199: Mux Trees

seldridge, 2019-10-01:
The current implementation of encoder/mux utilities all result in a completely flat sequence of muxes. This is both:

1. Counterintuitive because [the Scaladoc specifically talks about "mux tree" generation](https://github.com/freechipsproject/chisel3/blob/0289a2c59a7da690d9f46bc3003208b1bd7f2587/src/main/scala/chisel3/util/Mux.scala#L10) when this is a producing a very unbalanced tree.

2. Potentially sub-optimal because this is unbalanced and will result in worse reporting by tooling, e.g., Yosys will look at the long path and happily report that this is worse than Verilog `if`/`then` or `case`.

I do realize that FPGA and ASIC toolchains *should* do mux tree optimization, but it seems odd that generation of balanced mux trees isn't something that's natively supported (or maybe I'm just missing the right method).

[@aswaterman has previously expressed concern about doing mux tree optimization at the FIRRTL level because that breaks designer intent](https://github.com/freechipsproject/firrtl/issues/1049#issuecomment-471234634). I'm fine with that line of reasoning. However, it does seem odd that the standard library is producing these skinny, sub-optimal mux trees.

I would propose either:

1. Changing the standard library mux tree generation to emit balanced trees
2. Adding additional methods/an option to generate balanced trees

### Example

The following priority encoder with 8 options results in 8 muxes:

```scala
class Bar(n: Int) extends MultiIOModule {
  val in = IO(Input(UInt(n.W)))
  val sel = IO(Input(UInt(n.W)))
  val out = IO(Output(Bool()))

  out := PriorityMux(sel.asBools.zip(in.asBools))
}
```

(Resulting FIRRTL:)
```
circuit Bar : 
  module Bar : 
    input clock : Clock
    input reset : UInt<1>
    input in : UInt<8>
    input sel : UInt<8>
    output out : UInt<1>
    
    node _T = bits(sel, 0, 0) @[<pastie> 26:26]
    node _T_1 = bits(sel, 1, 1) @[<pastie> 26:26]
    node _T_2 = bits(sel, 2, 2) @[<pastie> 26:26]
    node _T_3 = bits(sel, 3, 3) @[<pastie> 26:26]
    node _T_4 = bits(sel, 4, 4) @[<pastie> 26:26]
    node _T_5 = bits(sel, 5, 5) @[<pastie> 26:26]
    node _T_6 = bits(sel, 6, 6) @[<pastie> 26:26]
    node _T_7 = bits(sel, 7, 7) @[<pastie> 26:26]
    node _T_8 = bits(in, 0, 0) @[<pastie> 26:41]
    node _T_9 = bits(in, 1, 1) @[<pastie> 26:41]
    node _T_10 = bits(in, 2, 2) @[<pastie> 26:41]
    node _T_11 = bits(in, 3, 3) @[<pastie> 26:41]
    node _T_12 = bits(in, 4, 4) @[<pastie> 26:41]
    node _T_13 = bits(in, 5, 5) @[<pastie> 26:41]
    node _T_14 = bits(in, 6, 6) @[<pastie> 26:41]
    node _T_15 = bits(in, 7, 7) @[<pastie> 26:41]
    node _T_16 = mux(_T_6, _T_14, _T_15) @[Mux.scala 47:69]
    node _T_17 = mux(_T_5, _T_13, _T_16) @[Mux.scala 47:69]
    node _T_18 = mux(_T_4, _T_12, _T_17) @[Mux.scala 47:69]
    node _T_19 = mux(_T_3, _T_11, _T_18) @[Mux.scala 47:69]
    node _T_20 = mux(_T_2, _T_10, _T_19) @[Mux.scala 47:69]
    node _T_21 = mux(_T_1, _T_9, _T_20) @[Mux.scala 47:69]
    node _T_22 = mux(_T, _T_8, _T_21) @[Mux.scala 47:69]
    out <= _T_22 @[<pastie> 26:7]
```

**Type of issue**: feature request | other enhancement

**Impact**: API modification

**Development Phase**: request | proposal

**Please tell us about your environment:**
    - version: `3.2.0-RC2`

mattvilim, 2020-04-05:
Why are the Chisel library muxes, encoders, decoders, etc. immediately lowered to 2x1 mux/logic cascades? Retaining the higher level structure in the IR would allow for direct emission of Verilog case statements during codegen. This applies to tree-based (priority) encoders/decoders too.

As it is, Design Compiler and other synthesis tools can't infer MUX_OP and priority logic which prohibits common synthesis optimizations.

Emitting balanced trees isn't even a proper fix in my mind since the synthesis tools don't always choose a tree structure.

aswaterman, 2020-04-06:
It's important to note that the output of DC's elaboration phase makes things look quite a bit worse than they actually are.  For example, Chisel/Firrtl emit a multiply-add as two Verilog statements (tmp = a*b; res = tmp+c), which DC elaboration turns into a separate multiply and add.  But later in synthesis (during datapath extraction), the operators are merged with a carry-save adder.

I don't know the mechanisms for its mux optimizations, but I know that DC synthesis often emits a mux tree that's structurally dissimilar to what Chisel/Firrtl emit, despite the impression that the elaboration output would suggest.  My guess is that's probably because it doesn't do its own mux restructuring until there's timing information, which of course is unavailable during elaboration.

Emitting balanced trees as an automagic Firrtl optimization is probably an anti-solution, because that ignores late-arriving signals.

grebe, 2020-04-06:
I agree that pattern matching in the backend has problems, so it seems natural to make it easy in the frontend to emit what you want. Should something like my [TreeReduce implementation](https://github.com/grebe/ofdm/blob/afd9cd404d6748185237d28edfca821b897642a3/src/main/scala/ofdm/TreeReduce.scala#L3) be part of the standard library? I think many people end up writing something like this at some point. You could make an implicit so you could call this like you call reduce, e.g. `list.treeReduce()` instead of `list.reduce()`.

seldridge, 2020-04-06:
@mattvilim wrote:
> Why are the Chisel library muxes, encoders, decoders, etc. immediately lowered to 2x1 mux/logic cascades? Retaining the higher level structure in the IR would allow for direct emission of Verilog case statements during codegen. This applies to tree-based (priority) encoders/decoders too.

The FIRRTL IR is trying to be as simple as possible, so there are only two mechanisms for specifying conditional updates: `when`/`elsewhen` and `Mux` (with the former lowered to `Mux` during compilation). **Note that code gen improvements, like case statements, and having a simple IR are both achievable goals**. (I had started working on case statement emission, but haven't had time to get back to that.)

There are two tools to help with this: (1) restructuring IR statements before emission (e.g., `firrtl.transforms.FlattenRegUpdate` which is flattening `Mux`es involved in register updates to make it possible to emit `if`/`else if`/`else` blocks) and (2) the use of `Annotation`s to encode metadata which we can't and don't want to express in the IR. I expect that these tools are sufficient o get `case` emission working.

However, it is a wide open question of where these approaches break down or if there is a better pattern. MLIR introduces "dialects" for adding operations, types, etc. without blowing up the IR. It's at least an architectural pattern to consider.

@aswaterman wrote:
> It's important to note that the output of DC's elaboration phase makes things look quite a bit worse than they actually are. For example, Chisel/Firrtl emit a multiply-add as two Verilog statements [...]

The original motivation of this issue was poor Yosys synthesis estimates (analogous to DC elaboration estimates) giving a designer false feedback on their design. I admit it's idyllic to hope for a tight design-synthesis feedback loop that is meaningfully correlated with FPGA or ASIC placement.

Two things may help here:

  - `case` statement emission (https://github.com/freechipsproject/chisel3/issues/1198)
  - `always @*` emission (https://github.com/freechipsproject/firrtl/issues/1340) which would allow for the above to emit using `if`/`else if`/`else` (and eventually `case`) as opposed to only ternary statements for combinational logic.

@grebe: Looks cool. I also forgot about the `Vec.reduceTree` method which does already exist.

jackkoenig, 2023-05-09:
MFC implements this in more canonical ways:
```scala
//> using scala "2.13.10"
//> using lib "org.chipsalliance::chisel::5.0.0-RC2"
//> using plugin "org.chipsalliance:::chisel-plugin::5.0.0-RC2"
//> using options "-unchecked", "-deprecation", "-language:reflectiveCalls", "-feature", "-Xcheckinit", "-Xfatal-warnings", "-Ywarn-dead-code", "-Ywarn-unused", "-Ymacro-annotations"

import chisel3._
import circt.stage.ChiselStage

class Foo extends Module {
  val in = IO(Input(Vec(8, UInt(8.W))))
  val idx = IO(Input(UInt(3.W)))
  val out = IO(Output(UInt(8.W)))

  out := in(idx)
}

object Main extends App {
  println(
    ChiselStage.emitSystemVerilog(
      gen = new Foo,
      firtoolOpts = Array("-disable-all-randomization", "-strip-debug-info")
    )
  )
}
```
This gives:
```verilog
// Generated by CIRCT firtool-1.40.0
module Foo(
  input        clock,
               reset,
  input  [7:0] in_0,
               in_1,
               in_2,
               in_3,
               in_4,
               in_5,
               in_6,
               in_7,
  input  [2:0] idx,
  output [7:0] out
);

  wire [7:0][7:0] _GEN = {{in_7}, {in_6}, {in_5}, {in_4}, {in_3}, {in_2}, {in_1}, {in_0}};
  assign out = _GEN[idx];
endmodule
```
Aggregate preservation makes it even nicer, but the unbalanced mux tree is no more.

closed 2023-05-09 by jackkoenig