chipsalliance/chisel #1207: not found: object chisel3

IAMAl, 2019-10-13:
I have installed "sbt" in my local PC followed this instruction (Ubuntu 18):
```
https://github.com/freechipsproject/chisel3/wiki/Installation-Preparation
```
and my code includes this, at first line;
```
import chisel3._
```
Then I meet an error of import;
```
not found: object chisel3
```
Does this mean "PATH" setting problem?

jackkoenig, 2019-10-13:
You need to include chisel3 as a dependency in your project. See [my answer on StackOverflow](https://stackoverflow.com/questions/40470153/is-there-a-simple-example-of-how-to-generate-verilog-from-chisel3-module/40474979#40474979) and the [chisel-template repo](https://github.com/freechipsproject/chisel-template)

ekiwi, 2019-10-18:
Please reopen this issue if jack's suggestion does not work for you.

closed 2019-10-18 by ekiwi